ARM-DOE/pyart #1255: Plotting negative elevation angles for RHI scans

scollis, 2022-08-31:
### Discussed in https://github.com/ARM-DOE/pyart/discussions/1253

<div type='discussions-op-text'>

<sup>Originally posted by **Ted-Whittock** August 31, 2022</sup>
Hello,

I have a bunch of RHI files from a mobile radar that was positioned along a mountain ridge, and so we have negative elevation angles to capture the precipitation over the surrounding lower terrain. (Iink to sample data: https://drive.google.com/drive/folders/19cQ3aS5qBt2gsQYjZu5cg14U_j7qt5J8?usp=sharing).

I have stitched 2 different azimuths together manually and added terrain heights below the radar (see attached image). However, I cannot figure out why it appears that the negative elevation angles (down to -1.4 deg) are not plotting. Is there a way to plot these negative elevations, or does Pyart always ignore them?

I am also aware that the y-axis on my attached plot is incorrect, but I wanted to address the issue of the negative elevation angles first.

Thanks,
Ted

[image]

</div>

scollis, 2022-08-31:
@jfigui @wolfidan have you looked at negative RHI angles in PyRad?

jfigui, 2022-09-06:
Hi @scollis,

We do have negative angles at MeteoSwiss but I never really paid attention. @wolfidan, can you have a quick look to that?

wolfidan, 2022-09-12:
@scollis 
Sorry for the delay, I'm currently in vacation, I will take a look on thursday though,
Cheers

jfigui, 2022-09-13:
Thanks a lot!

mgrover1, 2022-09-14:
@Ted-Whittock I took a look at your sample file, and noticed that within the data quality parameters (antenna transition), they flag the negative elevation angles... by default, we filter out these transitions in the `plot_rhi` function, but you can change this behavior by setting

```python
filter_transitions=False
```

for example, I used the following snippet

```python
import pyart
import matplotlib.pyplot as plt

radar = pyart.io.read("cfrad.20170203_160235.683_to_20170203_160247.799_DOW7_v245_s03_az65.00_RHI.nc")

display = pyart.graph.RadarDisplay(radar)

display = pyart.graph.RadarDisplay(radar)
display.plot_rhi("DBZHCC", cmap='pyart_HomeyerRainbow', filter_transitions=False, edges=False)
plt.ylim(-5, 10)
```

which resulted in the following image
[image]