ARM-DOE/pyart #1256: Missing parameters in NEXRAD level II data

westzhicanchen, 2022-08-31:
Hi everyone, 

I'm using PyART to analyze NEXRAD level II data but I couldn't find some parameters that I need. Could anyone pitch in here on how to find the value of these parameters? This issue seems consistent since I dug into multiple data and none of them have those parameters.

The parameters missing are **_instrument_parameters_** and **_radar_parameters_**. They are defined in CF/Radial data file format as the following:
[image]
[image]

When I take a look at the NEXRAD level II data using Pyart and one example dataset as the following, here is what I get with regard to these two parameters:

```python
import pyart
# loading NEXRAD data
NEXRADpath = './data/NEXRAD/nexrad/harvey/'
NEXRADname = "KHGX20170826_140853_V06"
radar = pyart.io.read_nexrad_archive(NEXRADpath + NEXRADname)
print(radar.instrument_parameters)
print(radar.radar_parameters)
```

```python
{'unambiguous_range': {'data': array([466000., 466000., 466000., ..., 117000., 117000., 117000.],
      dtype=float32), 'comments': 'Unambiguous range', 'long_name': 'Unambiguous range', 'units': 'meters', 'meta_group': 'instrument_parameters'}, 'nyquist_velocity': {'data': array([ 8.92,  8.92,  8.92, ..., 35.55, 35.55, 35.55], dtype=float32), 'comments': 'Unambiguous velocity', 'long_name': 'Nyquist velocity', 'units': 'meters_per_second', 'meta_group': 'instrument_parameters'}}
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-24-fbeae2882d33> in <module>()
      9 
     10 # show radar_parameters
---> 11 print(radar.radar_parameters)

AttributeError: 'Radar' object has no attribute 'radar_parameters'
```

As we see here, only limited parameters (i.e., unambiguous range and Nyquist velocity) are available from **_instrument_parameters_**, and no parameter is available from **_radar_parameters_**, any thoughts on where I might find them?

zssherman, 2022-08-31:
@westzhicanchen Thanks for the heads up, I'll take a look. Instrument parameters should populate, so i'll see if something is going wrong in the reader using those example files. I believe for radar_parameters, it seems those are being combined into instrument parameters for other readers:
https://github.com/ARM-DOE/pyart/blob/f923551e1fd8de72c8db6c76e6bc9913bf635644/pyart/io/uf.py#L248
not seeing it in the nexrad reader itself, I'll do more digging

mgrover1, 2022-09-01:
While the core data model in Py-ART is based on CfRadial, not all of the files we read are CfRadial and follow this convention (ex. NEXRAD's data format). The NEXRAD data format is a collection of individual messages, which we parse in the reader written in Py-ART, extracting what instrument parameters are available (ex. nyquist velocity).

Taking a look at the NEXRAD data documentation, I do not see these radar parameter fields as one of the fields within the messages. This might be more of a question for the data provider (NOAA).

westzhicanchen, 2022-09-02:
Hi everyone, 

Thank you all for the quick and detailed response. Any chance you would recommend some person/link that I might find useful when reaching out to the data provider NOAA and asking about this?

mgrover1, 2022-09-12:
You can find the contact us section on the NCEI website, where they listed this email [ncei.info@noaa.gov](mailto:ncei.info@noaa.gov)