ARM-DOE/pyart #1257: Hello,does pyart have the function of mapping two radars' Composite Reflectivity?

lsxlv, 2022-09-12:


zssherman, 2022-09-12:
@lsxlv Hello! We are currently in the process of getting a composite reflectivity function within Py-ART, once that's in. I believe if you have both radars with the generated composite field, you should be able to map both radars to the map_radars_to_grid function to map them both.

lsxlv, 2022-09-13:
@zssherman Thanks for answer! I have drawn the base reflectivity mosaic of the two radars with 'map_to_grid' function. This function is really good.
You mean that if I can draw the composite  reflectivity of two radars on Cartesian coordinate  seperately, then I can use 'map_to_grid' function to map these two composite  reflectivity together ？

mgrover1, 2022-09-13:
@lsxlv you can pass multiple radars to the `pyart.map.grid_from_radars` function, for example:

```python
gridded_radar = pyart.map.grid_from_radars((radar1, radar2), grid_shape, grid_limits))
```

see the docs here[grid_from_radars docs](https://arm-doe.github.io/pyart/API/generated/pyart.map.grid_from_radars.html#pyart.map.grid_from_radars)

We use this workflow one of the examples [grid from radars example](https://arm-doe.github.io/pyart/examples/mapping/plot_map_two_radars_to_grid.html#sphx-glr-examples-mapping-plot-map-two-radars-to-grid-py)