ARM-DOE/pyart #1260: ENH: Add a new example showing hydrometeor classification and QPE functionality

mgrover1, 2022-09-14:
At the recent ERAD tutorials, there were some great new tutorials put together by the meteoswiss/meteofrance team (@wolfidan @jfigui )

[Sample notebook](https://openradarscience.org/erad2022/notebooks/pyart/exercice1_swiss_thunderstorm.html)

It would be nice to have a concise example of this in our examples gallery!

mgrover1, 2022-10-04:
@wolfidan following up here - would this be something you would be interested in working together on?

wolfidan, 2022-10-11:
Hi @mgrover1 , sorry for the delay, Yes I will try to add this in the examples gallery as soon as possible

mgrover1, 2022-10-11:
I think splitting these up, and then adding a more detailed blog post bringing these together... I see the blog post section as more full-workflow focused

PR #1317 (2022-11-08): DOC: Added two pyart examples (VAD and hydrometeors)
Closes #1260 

Dear pyart team,

Sorry for the long delay, I had to wait for the green light to share some of the data and had some fieldwork to do...

Here are two new examples for the pyart gallery taken from the erad2022 open radar course.

**plot_vad.py:**
Compute VAD for every sweep of a volumetric scan, compute a vertical profile of U and V wind components, display the results

**plot_hydrometeors.py:**
Compute attenuation and KDP, read temperature from another source and add it to radar object, compute hydrometeors, display the result [... cut]
closed 2022-11-08 by zssherman