ARM-DOE/pyart #1266: Publish wheels on PyPI ?

neutrinoceros, 2022-09-21:
Hi, I know this is a stretch, but would it be possible to start publishing some wheels on PyPI ?
I am aware that binaries are available on conda, I'm asking because amr-pyart get installed on yt's CI and costs about 2min to build on each job.

I am not familiar with pyart, is there a technical reason why wheel installs cannot be supported ?

mgrover1, 2022-09-21:
Yeah - we can consider publishing individual wheels on PyPI! The main limitation has been the trmm_rsl package, which is a C package that is not available within Python.

mgrover1, 2022-09-21:
We would also need a wheel for each cython version/OS (ex. `arm_pyart-1.12.7.post18-cp310-cp310-macosx_11_0_arm64.whl`)

mgrover1, 2022-09-21:
@neutrinoceros we have started a PR moving toward this, any feedback or help there would be greatly appreciated!

neutrinoceros, 2022-09-21:
I will take a look tomorrow, thank you very much for putting it together so quickly !

closed 2022-09-26 by zssherman

zssherman, 2022-09-26:
@neutrinoceros We closed the issue, but just pinging, it looks like the build wheels are now available on PyPI!

mgrover1, 2022-09-26:
Thanks for all your help here @neutrinoceros !

neutrinoceros, 2022-09-26:
Yup I saw that, thank you very much for being so reactive on this ! We are already seeing the result as a net ~2min gain on all CI jobs at yt, that will easily save hours of CPU time a week !

PR #1267 (2022-09-26): ENH: Add cibuildwheel to build wheels
Closes #1266 
Closes #1224