ARM-DOE/pyart #1268: AttributeError: <''> object has no attribute '_autoscaleXon'

nkay28, 2022-09-22:
I have received this error with multiple objects, while trying to plot reflectivity from s3 bucket object. I think it is originating from the radar variable or the plot_ppi_map itself. 
1) `AttributeError: 'GeoAxes' object has no attribute '_autoscaleXon'`
2) `AttributeError: 'GeoAxesSubplot' object has no attribute '_autoscaleXon'`

I tried to make sure its not on my end but not a 100% sure. 

Here's one of the complete error for reference:

```
AttributeError                            Traceback (most recent call last)
<ipython-input-38-9e299f8058e6> in <module>
     20 display = pyart.graph.RadarMapDisplay(my_radar)
     21 fig = plt.figure(figsize = [10,8])
---> 22 display.plot_ppi_map('reflectivity', sweep = 0, resolution = 'c',
     23                     vmin = -8, vmax = 64, mask_outside = False,
     24                     cmap = pyart.graph.cm.NWSRef,

C:\Users\Nitin_Kumar\anaconda3\lib\site-packages\pyart\graph\radarmapdisplay.py in plot_ppi_map(self, field, sweep, mask_tuple, vmin, vmax, cmap, norm, mask_outside, title, title_flag, colorbar_flag, colorbar_label, ax, fig, lat_lines, lon_lines, projection, min_lon, max_lon, min_lat, max_lat, width, height, lon_0, lat_0, resolution, shapefile, shapefile_kwargs, edges, gatefilter, filter_transitions, embellish, raster, ticks, ticklabs, alpha, edgecolors, **kwargs)
    298         if norm is not None:  # if norm is set do not override with vmin/vmax
    299             vmin = vmax = None
--> 300         pm = ax.pcolormesh(x * 1000., y * 1000., data, alpha=alpha,
    301                            vmin=vmin, vmax=vmax, cmap=cmap,
    302                            edgecolors=edgecolors, norm=norm,

C:\Users\Nitin_Kumar\anaconda3\lib\site-packages\cartopy\mpl\geoaxes.py in wrapper(self, *args, **kwargs)
    308 
    309         kwargs['transform'] = transform
--> 310         return func(self, *args, **kwargs)
    311     return wrapper
    312 

C:\Users\Nitin_Kumar\anaconda3\lib\site-packages\cartopy\mpl\geoaxes.py in pcolormesh(self, *args, **kwargs)
   1559 
   1560         """
-> 1561         result = self._pcolormesh_patched(*args, **kwargs)
   1562         self.autoscale_view()
   1563         return result

C:\Users\Nitin_Kumar\anaconda3\lib\site-packages\cartopy\mpl\geoaxes.py in _pcolormesh_patched(self, *args, **kwargs)
   1657         self.update_datalim(corners)
   1658         self.add_collection(collection)
-> 1659         self.autoscale_view()
   1660 
   1661         ########################

C:\Users\Nitin_Kumar\anaconda3\lib\site-packages\cartopy\mpl\geoaxes.py in autoscale_view(self, tight, scalex, scaley)
    855                                             scalex=scalex, scaley=scaley)
    856         # Limit the resulting bounds to valid area.
--> 857         if scalex and self._autoscaleXon:
    858             bounds = self.get_xbound()
    859             self.set_xbound(max(bounds[0], self.projection.x_limits[0]),

AttributeError: 'GeoAxesSubplot' object has no attribute '_autoscaleXon'
```

mgrover1, 2022-09-22:
@nkay28 - can you run `conda list` to check which versions of packages you are running?

mgrover1, 2022-09-22:
Alright - I tracked down the issue here.

@nkay28 - you will need to use the latest version of cartopy (0.21.0) to use the latest version of matplotlib (3.6.0)

See the discussion here https://github.com/SciTools/cartopy/pull/2054

There were some attributes removed in the new version of matplotlib, which cartopy uses, which causes some issues here. Let's move this to a discussion, so other users are aware of these sorts of issues. We currently do **require** cartopy as a dependency, so we can't pin this in our requirements (but we should consider adding it since it used so much by our community)