ARM-DOE/pyart #1272: Issue with gate lat/lon calculation and radar location when reading Level 2 radar file

chudlerk, 2022-09-22:
Hello,

I am reading in the following radar file from the amazon sever:
[https://noaa-nexrad-level2.s3.amazonaws.com/2009/10/14/KDAX/KDAX20091014_091934_V03.gz](url)

When trying to plot the radar data on a map, I'm noticing some weird results for the latitude and longitude values returned with `get_gate_lat_lon_alt`:

```
radar = pyart.io.read('C:/Users/kchudler/Downloads/KHNX20060724_020338.gz')
lat, lon, alt = radar.get_gate_lat_lon_alt(0)
print('Lon Range:',lon.max(), lon.max())
print('Lat Range:',lat.max(), lat.max())
Lon Range: 4.13013044962621 4.13013044962621
Lat Range: 4.130129685785962 4.130129685785962
```

After doing some digging, I think it has to do with the fact that the radar location, as stored in `radar.longitude` and `radar.latitude` after reading in the file are incorrect (data value is 0):
print(radar.longitude)
{'long_name': 'Longitude', 'standard_name': 'Longitude', 'units': 'degrees_east', 'data': array([0.])}

print(radar.latitude)
{'long_name': 'Latitude', 'standard_name': 'Latitude', 'units': 'degrees_north', 'data': array([0.])}

It could just be that the data is simply stored wrong in the original level 2 file...in which case maybe there is nothing that can be done. However, other radar data plotters (i.e. [](https://radar.quadweather.com/)) seem to be able to read/plot the data just fine on a map, so there should be some way to do it...

mgrover1, 2022-09-23:
@chudlerk which version of Py-ART are you using?

I tried the following which appeared to work (with the latest version Py-ART 1.13.0):

```python
radar = pyart.io.read("KDAX20091014_091934_V03.gz")
print(radar.latitude['data'], radar.longitude['data'])
[38.50111008] [-121.67783356]
```

Or, reading directly from Amazon:

```python
radar = pyart.io.read_nexrad_archive("s3://noaa-nexrad-level2/2009/10/14/KDAX/KDAX20091014_091934_V03.gz")
print(radar.latitude['data'], radar.longitude['data'])
[38.50111008] [-121.67783356]
```

chudlerk, 2022-09-23:
Whoops, I pasted the wrong URL in my question. I've updated it with one that doesn't work. I'm having trouble installing 1.13.0 with anaconda, but it doesn't work on 1.12.2

mgrover1, 2022-09-23:
What sort of conda installation issues are you running into?

mgrover1, 2022-09-23:
@chudlerk - the older NEXRAD data is missing the required metadata (message 31). See this discussion https://github.com/ARM-DOE/pyart/discussions/1174 where we mention the ability to manually specify which station the radar data is from, which fixes this issue.

```python
radar = pyart.io.read_nexrad_archive("s3://noaa-nexrad-level2/2006/07/24/KHNX/KHNX20060724_020338.gz", station="KHNX")
print(radar.latitude['data'], radar.longitude['data'])
```