ARM-DOE/pyart #1293: radardisplay does not recognize c-sapr "sector" plot type

mvanlierwalq, 2022-10-20:
tryping to plot the csapr file houcsapr2cfrS2.a1.20220617.190001.nc, display.plot throws an error owing to unknown plot type "sector" : `ValueError: unknown scan_type sector`
can this be added to pyart?

mgrover1, 2022-10-20:
Yeah... currently "sector" is not one of these options, but we can consider adding it! Treating it the same as a PPI... which I think would be the right assumption @mvanlierwalq

zssherman, 2022-10-20:
Yeah agreed, it should be a simple fix

mvanlierwalq, 2022-10-20:
@zssherman thanks! will have to wait until tonight or tomorrow, having trouble connecting to my computer. will let you know.

mgrover1, 2022-10-28:
@mvanlierwalq any updates here? Did the fix work?

zssherman, 2022-11-07:
fixed in #1294 can be reopened if need be.

closed 2022-11-07 by zssherman