ARM-DOE/pyart #1305: Issue with echo_class.py

swatelet, 2022-10-28:
Hi all,

I noticed that the line 157 (if 'frequency' in radar.instrument_parameters:) of echo_class.py generates this error when instrument_parameters of a radar object is set to None:

"TypeError: argument of type 'NoneType' is not iterable"

I would suggest to add a check of None on line 157, what do you think?

Cheers,
Sylvain

mgrover1, 2022-10-28:
Hi @swatelet - thanks for catching this. I think that sounds like a reasonable solution. Would you be interested in implementing that fix?

swatelet, 2022-10-28:
Thanks but unfortunately I have no time for that right now, but feel free to do it.

zssherman, 2022-10-28:
I'll take a crack at this shortly.

mgrover1, 2022-10-28:
Closed by #1306

closed 2022-10-28 by zssherman