ARM-DOE/pyart #1328: wrong gate_size when reading IRIS/Sigmet files (?)

aladinor, 2022-11-21:
Hi everyone, 

Following up [issue#51](https://github.com/openradar/xradar/issues/51) I tried to read an iris file `xradar` and I got a dimension error because of the bin size. Metadata under `xradar` gives us 300 m (30000cm) in bin size. However, when reading the same file using `pyart` the bin size turns out to be 299.0 m. 

After a revision @kmuehlbauer suggested that there might be an issue when reading the bin size from the raw data using `pyart` [issue51#comment](https://github.com/openradar/xradar/issues/51#issuecomment-1321102939).  This is a snippet of my code

```
import pyart
import fsspec
import xarray as xr
import xradar as xd
import boto3
import botocore
from botocore.client import Config
from datetime import datetime

def create_query(date, radar_site):
    """
    Creates a query for listing IDEAM radar files stored in AWS bucket
    :param date: date to be queried. e.g datetime(2021, 10, 3, 12). Datetime python object
    :param radar_site: radar site e.g. Guaviare
    :return: string with AWS bucket structure
    """
    prefix = f'l2_data/{date:%Y}/{date:%m}/{date:%d}/{radar_site}/{radar_site[:3].upper()}{date:%y%m%d}'
    return prefix

def main():
    str_bucket = 's3://s3-radaresideam/'
    s3 = boto3.resource('s3',
                        config=Config(signature_version=botocore.UNSIGNED, user_agent_extra='Resource'))
    bucket = s3.Bucket('s3-radaresideam')
    query = create_query(date=datetime(2021, 10, 3, 12), radar_site='Guaviare')
    radar_files = [f'{str_bucket}{i.key}' for i in bucket.objects.filter(Prefix=f"{query}")]

    # using context manager for reading radar data from guaviare radar
    of = pyart.io.prepare_for_read(radar_files[2], storage_options={'anon': True})
    with of as f:
        radar = pyart.io.read(f)
        print(radar.range['meters_between_gates'])
        f.close()
```

We would like to know your thoughts. 

Cheers, 

Alfonso

kmuehlbauer, 2022-11-21:
I think the problematic lines are:

https://github.com/ARM-DOE/pyart/blob/679e18650b205d7e2bfb87a503770058eff3786e/pyart/io/sigmet.py#L281-L283

mgrover1, 2022-11-21:
@kmuehlbauer / @aladinor good catch! Thanks for filing this issue.

closed 2023-02-12 by aladinor