ARM-DOE/pyart #1333: NEXRAD Non-Reflectivity Values

mgrover1, 2022-11-22:
Building on the discussion here - https://openradar.discourse.group/t/reflectivity-and-velocity-resolution-from-aws/97

The main question is whether we are parsing NEXRAD Level 2 files with the proper digital resolution.

Reflectivity should be at 8 bit resolution (every 0.5 dBZ), but other fields such as radial velocity and spectrum width should be higher? With differences in values on the order of 0.1 m/s, etc.

@dopplershift do you know where to find this information so we can verify we are treating this properly? Or do you know if the current implementation in MetPy/Py-ART handle this properly?

dopplershift, 2022-11-22:
So different fields within the Level2 data do have different precisions and amounts of bits actually used. That information is in [this ICD](https://www.roc.noaa.gov/wsr88d/WSR88D/PublicDocs/ICDs/2620002V.pdf) (which seems to be 404-ing on their site at the moment):

[image]

So PhiDP and ZDR are stored in 16-bit ints, having 10 and 11 bits of precision, respectively. From the post you linked, I also think there is some confusion. Level 2 vel and reflectivity have 8-bits and due to the limited range, that corresponds to 0.5 increments--which is represented exactly in floating point. Correlation coefficient appears to have more digits, but it's still just 8-bit data.

#903 was trying to accommodate properly for the different data word sizes. I know MetPy gets this right.

mgrover1, 2022-11-22:
Thanks @dopplershift !! I spent quite a bit of time trying to track down that table... 

And since that is the case with the different bit percisions, then Py-ART gets this right too (since we follow the same approach y'all did)

Thanks for the quick response here. I am going to repost this on the Open Radar discourse as well to respond to Brenda's discussion.