ARM-DOE/pyart #1336: Error with gamic file having different bin numbers depending on elevation

tsbiscaro, 2022-11-24:
I am reading a GAMIC file with 600 gates in the first 5 elevations and 800 in the last 3. This results in an error as below:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tbiscaro/anaconda3/envs/radar/lib/python3.8/site-packages/pyart/aux_io/gamic_hdf5.py", line 204, in read_gamic
    field_dic['data'] = gfile.moment_data(group, 'float32')
  File "/home/tbiscaro/anaconda3/envs/radar/lib/python3.8/site-packages/pyart/aux_io/gamicfile.py", line 142, in moment_data
    data[start:end+1, :sweep_data.shape[1]] = sweep_data[:]
  File "/home/tbiscaro/anaconda3/envs/radar/lib/python3.8/site-packages/numpy/ma/core.py", line 3387, in __setitem__
    _data[indx] = dval
ValueError: could not broadcast input array from shape (361,800) into shape (361,600)

The code is using the first sweep dimension (361, 600) to all sweeps, leading to this error when a sweep have a different azimuth/bin count than the first sweep. This is defined in gamicfile.py line 135.

ngates = int(self._hfile['/scan0/how'].attrs['bin_count'])

kmuehlbauer, 2022-11-24:
@tsbiscaro You might not get an answer today  🦃 .

So the reader would need to get the max bin number and use this for the creation of the array.

But this only works if the range resolution  is same in all sweeps.

If you only want to have a look at the data you might consider trying out

https://docs.openradarscience.org/projects/xradar/en/latest/

kmuehlbauer, 2022-11-24:
@tsbiscaro You might try the solution proposed in #1337. It's not as geeky as the PR number suggests, though.

tsbiscaro, 2022-11-24:
Thank you! Didn't know about xradar, I will take a look at it.

zssherman, 2022-11-26:
Fixed in #1337, closing

closed 2022-11-26 by zssherman