ARM-DOE/pyart #1340: Possible issue with region based dealiasing

mgrover1, 2022-12-05:
When I was going through and adding pre-commit hooks, I noticed the following issue in the region based dealiasing module

The `vel_wo_cur` variable is never used...

https://github.com/ARM-DOE/pyart/blob/main/pyart/correct/region_dealias.py#L463

Should we be setting that value to some array? Do we need to store it somewhere? It was changed by @rcjackson last year I think

mgrover1, 2022-12-05:
Here is the commit that changed the behavior here 
https://github.com/ARM-DOE/pyart/commit/7485d867cbf4d726b11c3bc95206da74bd4084f4

rcjackson, 2022-12-05:
It’s not used anymore but left over there on accident. The references can be removed since they are not used.

Bobby

https://github.com/ARM-DOE/pyart/blob/main/pyart/correct/region_dealias.py#L463

closed 2022-12-05 by mgrover1