ARM-DOE/pyart #1345: Trouble Reading Chill Radar Data

mgrover1, 2022-12-06:
Opening a new issue, based on a question from @melovato

```
I am getting the same "error: unpack requires a buffer of 48 bytes" when reading CSU-CHILL CHL data. I have tried both pyart.io.read(filename) and pyart.io.ready_chl(filename) and the same error occurs. I am using Python 3.9 and arm_pyart 1.14.1 that I installed via conda. I can provide the package list and any other information if needed.

Thank you,
Mateo

```

_Originally posted by @melovato in https://github.com/ARM-DOE/pyart/issues/826#issuecomment-1336572807_

Can you share a data file that we can troubleshoot with?

mgrover1, 2022-12-06:
Can you upload to google drive, and share with my email mgroverwx@gmail.com?

melovato, 2022-12-06:
Done, let me know if you have any issues.

mgrover1, 2022-12-07:
@melovato is this happening with all of the chill files you are working with, or just a single file?

mgrover1, 2022-12-07:
It looks like there is a mismatch in the ray header within the file. For Chill files, that part of the dataset should include the following structure:

```python
(('azimuth', 'f'),
 ('elevation', 'f'),
 ('azimuth_width', 'f'),
 ('elevation_width', 'f'),
 ('gates', 'H'),
 ('beam_index', 'H'),
 ('ns_time', 'I'),
 ('time', 'Q'),
 ('bit_mask', 'Q'),
 ('ray_number', 'I'),
 ('num_pulses', 'I'))`
```

Which corresponds to a buffer size of 48 bytes.

Within the file you provided, that part of the dataset  (the ray header section) is 64 bytes, indicating there is additional information included, but without knowing what additional fields are there, we cannot determine the expected buffer size.