ARM-DOE/pyart #1371: Globally enabling DeprecationWarnings

dopplershift, 2023-01-06:
I was trying to figure out why some code I had in a notebook to silence deprecation warnings (we've got a few coming from stdlib, urllib3, and a ton from cartopy+shapely v2), and I couldn't figure out why it wasn't working. Then I stumbled across:

https://github.com/ARM-DOE/pyart/blob/0714f46a8258f043b2ab9ab59d863b4095d1a1ae/pyart/__init__.py#L28

This is changing global state for every user and enabling every deprecation warning from every library, overriding default Python behavior for everyone. I know the original commit said this was common in the scientific stack, but I just did a full import of numpy, matplotlib, scipy, pandas, xarray, and the list of installed warning filters would indicate PyART is the only library setting this.

For CI/CD, the standard these days is to manually enable them for test runs. If the concern is making sure users see all of your warnings, you can use `filterwarnings()` instead of `simplefilter()`, which would let you specify a module regex so that the filter only applies to warnings from pyart modules.

zssherman, 2023-01-06:
@dopplershift I agree, I can change the behavior so this no longer occurs. With most packages, how do they normally go about printing deprecation warnings? Do you think its best for us to just remove this? I guess I'm more trying to wrap my head around it still

mgrover1, 2023-01-07:
Thanks for bringing this up @dopplershift - we can remove that.

PR #1372 (2023-01-07): Remove deprecation line in init file
Change how we automatically deal with warnings

Closes #1371
closed 2023-01-07 by zssherman