ARM-DOE/pyart #1400: No colorbar_orient option in pyart.graph.RadarMapDisplay.plot_ppi_map()

joshua-hampton, 2023-03-16:
The `colorbar_orient` option that is available in `pyart.graph.RadarDisplay.plot_ppi()` is not available in `pyart.graph.RadarMapDisplay.plot_ppi_map()` (or `pyart.graph.RadarMapDisplayBasemap.plot_ppi_map()`). Is this intentional or just an oversight? If the latter, I have a fork that I'm happy to submit a PR from with this option added in.

zssherman, 2023-03-16:
@joshua-hampton Hello! I'm honestly not too sure. It could have been an oversight. If your willing to do a PR that would be awesome! I would check to see if the plots look reasonable. My only theory if it wasn't included for a reason would be it not playing well cartopy geoaxes, but honestly not sure. So I would say its probably fine to go ahead with a PR if the behavior seems to be reasonable.

joshua-hampton, 2023-03-16:
@zssherman Thanks, I'll put in that PR. Here's a quick plot from our X-band, as far as I can see there's nothing odd going on (with the colour bar at least).
```
import pyart
import cartopy.crs as ccrs
import matplotlib.pyplot as plt

radar_file = 'ncas-mobile-x-band-radar-1_chilbolton_20170517-113106_SUR_v1.nc'
radar = pyart.io.read(radar_file)
display = pyart.graph.RadarMapDisplay(radar)
fig = plt.figure(figsize = (10,8))
ax = plt.subplot(111, projection=ccrs.PlateCarree())
display.plot_ppi_map('dBZ', ax=ax, colorbar_orient='horizontal', resolution = '10m')
gl = ax.gridlines(crs=ccrs.PlateCarree(),
                  draw_labels=True,
                  linewidth=1, 
                  color='gray', 
                  alpha=0.3,
                  linestyle='--')
gl.top_labels = False
gl.right_labels = False
plt.show()
```

[image]

zssherman, 2023-03-16:
@joshua-hampton That looks reasonable to me! And sounds good thank you!

PR #1401 (2023-03-16): colorbar_orient option to plot_ppi_map functions

- [x] Closes #1400
- [x] Documentation reflects changes

Adds `colorbar_orient` as an option to `pyart.graph.RadarMapDisplay.plot_ppi_map()` and `pyart.graph.RadarMapDisplayBasemap.plot_ppi_map()`.
Documentation changed to reflect this in as far as adding bits in the function docstrings, I don't know if any more than that is needed.
closed 2023-03-16 by mgrover1