ARM-DOE/pyart #1408: BUG: read_odim_hdf5 - TypeError: 'numpy.float32' object cannot be interpreted as an integer

Vforcell, 2023-03-29:
* Py-ART version: 1.14.6
* Python version: 3.11 (same with 3.10)
* Operating System: Linux / Ubuntu

### Description

An error is raised when using `pyart.aux_io.odim_hdf5.read_odim_hdf5`:

`TypeError: 'numpy.float32' object cannot be interpreted as an integer`

The line to blame is
https://github.com/ARM-DOE/pyart/blob/75cb88466a74b48680c0cdf1400e3ed8a92d35f1/pyart/aux_io/odim_h5.py#L333

The `datetime.datetime.utcfromtimestamp` function doesn't seem to like `numpy.float32` dtype anymore.

A quick fix (that worked for me) is to specify "float64" in the dtype of  `t_data`, instead of "float32":

https://github.com/ARM-DOE/pyart/blob/75cb88466a74b48680c0cdf1400e3ed8a92d35f1/pyart/aux_io/odim_h5.py#L327

mgrover1, 2023-03-29:
Do you have a sample file that could be used here? For testing?

Vforcell, 2023-03-30:
[radar_file.zip](https://github.com/ARM-DOE/pyart/files/11109084/radar_file.zip)

Vforcell, 2023-04-17:
@mgrover1 This ODIM `.hdf5` file has been produced by the Météo-France quality control chain.

It might be that there is something with the file and the way it encodes time. But I don't see anything wrong on that part.

I think the problem is that `float32` type has been somehow deprecated for `datetime.datetime.utcfromtimestamp`, or it messes up with it.

kmuehlbauer, 2023-04-17:
@Vforcell My guess, too, is that it's connected to the dtype. ODIM_H5 is referencing these values as `array of double` so `float64` should work. Maybe this just surfaces now, where more systems really use 64bit resolution.

One special thing is still to this file as it contains a compound type dataset (`legend`) attached to `data7`. This part cannot be read by netCDF4, which has to kept in mind when exploring this via xarray or the like.

tealmill, 2023-05-03:
Went down a little rabbit hole investigating this myself and it turns out the [ODIM spec](https://www.eumetnet.eu/wp-content/uploads/2019/01/ODIM_H5_v23.pdf) specifically declares these parameters as double aka float64, so an archive with float32 in there is technically malformed/not conformant with the specification. I don't know what's the policy of pyart regarding compatibility versus conformity but the format the data is being saved in should be probably changed.

tealmill, 2023-05-03:
ODIM spec explicitly states the time of the startazT and stopazT must be a double aka float64. The [array initialization here](https://github.com/ARM-DOE/pyart/blob/52f1e1ba39f73d8db48aa6d4ea0f9ea9c3a9746b/pyart/aux_io/odim_h5.py#L327) will coerce it into an array of float32. This seems like a fairly straightforwardly a bug, that array should be initialized as float64 and changing it to such fixes this error.

mgrover1, 2023-05-03:
@Vforcell testing on that file to make sure things are fixed before merging

PR #1425 (2023-05-03): FIX: Fix type of time for odim reader
- [x] Closes #1408 
- [x] Documentation reflects changes
closed 2023-05-03 by mgrover1