ARM-DOE/pyart #1426: vad_michelson produces strange and incorrect results on Py-ART 1.14.2 and above

CAM-Gerlach, 2023-05-12:
* Py-ART version: 1.14.2 - 1.15.0
* Python version: 3.11.3
* Operating System: Windows 10

### Description

In in Py-ART 1.14.2 and above, up to and including 1.15.0, the `pyart.retrieve.vad_michelson` function produces strange and apparently incorrect results, with much too high and variable wind speeds, on both raw and corrected NEXRAD and MAX data, including some with indicated wind speeds on the order of many hundreds to even thousands of meters per second.

By contrast, in Py-ART 1.14.1 and below, the VAD wind profiles match the actual data and the expected results much more closely. I don't see where such an intentional change is documented in the release notes, nor does this appear to be the same issue as that discussed in #992 (given it only occurred recently, and results in the values being too high and variable rather than too low.

We've confirmed this on multiple files, including [this sample one in the reproducer below](https://github.com/ARM-DOE/pyart/files/11457360/cfradial.20080822_1502_VolScan_KJAX_SUR.nc.zip) and the sole change to the environment that causes the problem to reproduce or not is the Py-ART version, rather than any dependencies or other changes. This was tested in a brand new conda environment just freshly created today with only Py-ART (and Spyder-kernels) installed.

### What I Did

Here's a minimal reproducer that produces drastically different results on Py-ART 1.14.1 vs. 1.14.2 on the same file (a KJAX WSR-88D volume scan, in this instance, which I've attached)

```python
import pyart
import matplotlib.pyplot as plt
import numpy as np

radar = pyart.io.read('cfradial.20080822_1502_VolScan_KJAX_SUR.nc')
vad = pyart.retrieve.vad_michelson(radar, vel_field='velocity')

fig = plt.figure(figsize=(5,8))
ax = fig.add_subplot(111)
ax.plot(vad.speed, vad.height)
plt.title('KJAX VAD from 22 Aug. 2008 at 1502 Z - Velocity')
ax.set_xlabel('Wind Speed [m/s]')
ax.set_ylabel('Height [AGL]')
ax.set_xlim(0, np.nanmax(vad.speed) + 2)
```

### Expected/Actual

| Expected (Py-ART 1.14.1) | Actual (Py-ART 1.14.2) |
| --- | --- |
| [image] | [image] |

To note, the printed output from running the function is the same as the Expected case, for what that's worth. It was only the returned values that were different.

### Conda list output

<details>

<summary>Conda list output of an affected environment</summary>

```
# packages in environment at C:\Users\Owner\miniforge3\envs\pyart:
#
# Name                    Version                   Build  Channel
aiobotocore               2.5.0              pyhd8ed1ab_0    conda-forge
aiohttp                   3.8.4           py311ha68e1ae_0    conda-forge
aioitertools              0.11.0             pyhd8ed1ab_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
arm_pyart                 1.15.0          py311ha68e1ae_1    conda-forge
asttokens                 2.2.1              pyhd8ed1ab_0    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
blosc                     1.21.3               hdccc3a2_0    conda-forge
botocore                  1.29.76            pyhd8ed1ab_0    conda-forge
brotli                    1.0.9                hcfcfb64_8    conda-forge
brotli-bin                1.0.9                hcfcfb64_8    conda-forge
brotlipy                  0.7.0           py311ha68e1ae_1005    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2023.5.7             h56e8100_0    conda-forge
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
[... 120 lines cut ...]
ucrt                      10.0.22621.0         h57928b3_0    conda-forge
urllib3                   1.26.15            pyhd8ed1ab_0    conda-forge
vc                        14.3                hb25d44b_16    conda-forge
vc14_runtime              14.34.31931         h5081d32_16    conda-forge
versioneer                0.28               pyhd8ed1ab_0    conda-forge
vs2015_runtime            14.34.31931         hed1258a_16    conda-forge
wcwidth                   0.2.6              pyhd8ed1ab_0    conda-forge
wheel                     0.40.0             pyhd8ed1ab_0    conda-forge
win_inet_pton             1.1.0              pyhd8ed1ab_6    conda-forge
wrapt                     1.15.0          py311ha68e1ae_0    conda-forge
xarray                    2023.4.2           pyhd8ed1ab_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yarl                      1.9.1           py311ha68e1ae_0    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zipp                      3.15.0             pyhd8ed1ab_0    conda-forge
zstd                      1.5.2                h12be248_6    conda-forge
```

</details>

Version 1.14.2-py311ha68e1ae_0 was also affected, while 1.14.1-py311ha68e1ae_0 was confirmed okay.

Let us know if there's anything else you need from our end. Thanks!

zssherman, 2023-05-12:
@CAM-Gerlach Hmm, yeah that's very interesting. As you stated, yeah nothing has been changed besides a warning removal. I'm wondering if there is a dependency issue or a dependency updated how nans or masked values are being handled and is causing issues... @mgrover1 thoughts?

mgrover1, 2023-05-12:
Yeah, I suspect it is to do with with how the booleans (masked values) are being handled. I will dig into this.

mgrover1, 2023-05-17:
Thank you for catching this @CAM-Gerlach - during our formatting changes, a logical argument was changed and caused this issue... a fix is implemented in #1428 and we will be sure to cut a new release with this bug fix.

closed 2023-05-17 by mgrover1

CAM-Gerlach, 2023-05-17:
Thanks for the fix!

PR #1428 (2023-05-17): Fix masking issue in VAD michelson
- [x] Closes #1426 
- [x] Tests added
- [x] Documentation reflects changes