ARM-DOE/pyart #1429: plot_ppi_map() throwing "QuadMesh.set() got an unexpected keyword argument 'add_grid_lines'"

kylejgillett, 2023-05-20:
* Py-ART version: 1.15.0
* Python version: 3.9.16
* Operating System: Windows 

### Description
Since recently updating pyart, I am getting the attribute error "`QuadMesh.set()` got an unexpected keyword argument '`add_grid_lines`'" when trying to use the same `plot_ppi_map()` I had been using prior to updating. I am attempting to set `add_grid_lines` to false. 

Just a side note, I have also been making updates to other packages recently too, not sure if that would be involved. 

I see in the traceback that `add_grid_lines` is not included in the kwargs, but it's still up on the docs: https://arm-doe.github.io/pyart/API/generated/pyart.graph.RadarMapDisplay.plot_ppi_map.html#pyart.graph.RadarMapDisplay.plot_ppi_map

### What I Did

**CODE:**
```
display.plot_ppi_map('reflectivity',
                     sweep=0,
                     ax=map_ax,
                     vmin=-20,
                     title_flag = False,
                     colorbar_flag = False,
                     cmap='pyart_LangRainbow12',
                     resolution='10m',
                     gatefilter=gatefilter,
                     vmax=70,
                     lat_lines=None, 
                     lon_lines=None,
                     add_grid_lines=False,
                     zorder=2)

```

**TRACEBACK:**

```
AttributeError                            Traceback (most recent call last)
Cell In[52], line 6
      4 gatefilter.exclude_equal('reflectivity', 1)
      5 display = pyart.graph.RadarMapDisplay(radar)
----> 6 display.plot_ppi_map('reflectivity',
      7                      sweep=0,
      8                      ax=map_ax,
      9                      vmin=-20,
     10                      title_flag = False,
     11                      colorbar_flag = False,
     12                      cmap='pyart_LangRainbow12',
     13                      resolution='10m',
     14                      gatefilter=gatefilter,
     15                      vmax=70,
     16                      lat_lines=None, 
     17                      lon_lines=None,
     18                      add_grid_lines=False,
     19                      zorder=2)
     21 plt.show()

File ~\miniconda3\envs\oldNumpy2\lib\site-packages\pyart\graph\radarmapdisplay.py:296, in RadarMapDisplay.plot_ppi_map(self, field, sweep, mask_tuple, vmin, vmax, cmap, norm, mask_outside, title, title_flag, colorbar_flag, colorbar_label, ax, fig, lat_lines, lon_lines, projection, min_lon, max_lon, min_lat, max_lat, width, height, lon_0, lat_0, resolution, shapefile, shapefile_kwargs, edges, gatefilter, filter_transitions, embellish, raster, ticks, ticklabs, alpha, edgecolors, **kwargs)
    294 if norm is not None:  # if norm is set do not override with vmin/vmax
    295     vmin = vmax = None
--> 296 pm = ax.pcolormesh(x * 1000., y * 1000., data, alpha=alpha,
    297                    vmin=vmin, vmax=vmax, cmap=cmap,
    298                    edgecolors=edgecolors, norm=norm,
    299                    transform=self.grid_projection, **kwargs)
    301 # plot as raster in vector graphics files
    302 if raster:

File ~\miniconda3\envs\oldNumpy2\lib\site-packages\cartopy\mpl\geoaxes.py:318, in _add_transform.<locals>.wrapper(self, *args, **kwargs)
    313     raise ValueError(f'Invalid transform: Spherical {func.__name__} '
    314                      'is not supported - consider using '
    315                      'PlateCarree/RotatedPole.')
    317 kwargs['transform'] = transform
--> 318 return func(self, *args, **kwargs)

File ~\miniconda3\envs\oldNumpy2\lib\site-packages\cartopy\mpl\geoaxes.py:1785, in GeoAxes.pcolormesh(self, *args, **kwargs)
   1782 # Add in an argument checker to handle Matplotlib's potential
   1783 # interpolation when coordinate wraps are involved
   1784 args, kwargs = self._wrap_args(*args, **kwargs)
-> 1785 result = super().pcolormesh(*args, **kwargs)
   1786 # Wrap the quadrilaterals if necessary
   1787 result = self._wrap_quadmesh(result, **kwargs)

File ~\miniconda3\envs\oldNumpy2\lib\site-packages\matplotlib\__init__.py:1442, in _preprocess_data.<locals>.inner(ax, data, *args, **kwargs)
   1439 @functools.wraps(func)
   1440 def inner(ax, *args, data=None, **kwargs):
   1441     if data is None:
-> 1442         return func(ax, *map(sanitize_sequence, args), **kwargs)
   1444     bound = new_sig.bind(ax, *args, **kwargs)
   1445     auto_label = (bound.arguments.get(label_namer)
   1446                   or bound.kwargs.get(label_namer))

File ~\miniconda3\envs\oldNumpy2\lib\site-packages\matplotlib\axes\_axes.py:6229, in Axes.pcolormesh(self, alpha, norm, cmap, vmin, vmax, shading, antialiased, *args, **kwargs)
   6225     C = C.ravel()
   6227 kwargs.setdefault('snap', mpl.rcParams['pcolormesh.snap'])
-> 6229 collection = mcoll.QuadMesh(
   6230     coords, antialiased=antialiased, shading=shading,
   6231     array=C, cmap=cmap, norm=norm, alpha=alpha, **kwargs)
   6232 collection._scale_norm(norm, vmin, vmax)
   6234 coords = coords.reshape(-1, 2)  # flatten the grid structure; keep x, y

File ~\miniconda3\envs\oldNumpy2\lib\site-packages\matplotlib\collections.py:1939, in QuadMesh.__init__(self, coordinates, antialiased, shading, **kwargs)
   1936 self._bbox.update_from_data_xy(self._coordinates.reshape(-1, 2))
   1937 # super init delayed after own init because array kwarg requires
   1938 # self._coordinates and self._shading
-> 1939 super().__init__(**kwargs)
   1940 self.set_mouseover(False)

File ~\miniconda3\envs\oldNumpy2\lib\site-packages\matplotlib\_api\deprecation.py:454, in make_keyword_only.<locals>.wrapper(*args, **kwargs)
    448 if len(args) > name_idx:
    449     warn_deprecated(
    450         since, message="Passing the %(name)s %(obj_type)s "
    451         "positionally is deprecated since Matplotlib %(since)s; the "
    452         "parameter will become keyword-only %(removal)s.",
    453         name=name, obj_type=f"parameter of {func.__name__}()")
--> 454 return func(*args, **kwargs)

File ~\miniconda3\envs\oldNumpy2\lib\site-packages\matplotlib\collections.py:201, in Collection.__init__(self, edgecolors, facecolors, linewidths, linestyles, capstyle, joinstyle, antialiaseds, offsets, offset_transform, norm, cmap, pickradius, hatch, urls, zorder, **kwargs)
    198 self._offset_transform = offset_transform
    200 self._path_effects = None
--> 201 self._internal_update(kwargs)
    202 self._paths = None

File ~\miniconda3\envs\oldNumpy2\lib\site-packages\matplotlib\artist.py:1223, in Artist._internal_update(self, kwargs)
   1216 def _internal_update(self, kwargs):
   1217     """
   1218     Update artist properties without prenormalizing them, but generating
   1219     errors as if calling `set`.
   1220 
   1221     The lack of prenormalization is to maintain backcompatibility.
   1222     """
-> 1223     return self._update_props(
   1224         kwargs, "{cls.__name__}.set() got an unexpected keyword argument "
   1225         "{prop_name!r}")

File ~\miniconda3\envs\oldNumpy2\lib\site-packages\matplotlib\artist.py:1197, in Artist._update_props(self, props, errfmt)
   1195             func = getattr(self, f"set_{k}", None)
   1196             if not callable(func):
-> 1197                 raise AttributeError(
   1198                     errfmt.format(cls=type(self), prop_name=k))
   1199             ret.append(func(v))
   1200 if ret:

AttributeError: QuadMesh.set() got an unexpected keyword argument 'add_grid_lines'

```

mgrover1, 2023-05-22:
@kylejgillett Thanks for letting us know! I am unable to reproduce this error - can you run a `conda list` command to double check with versions of packages are being used here?

zssherman, 2023-09-22:
Hello @kylejgillett, were you still encountering this issue?

kylejgillett, 2023-09-26:
Hey! Apologies for the delayed response. I had to completely rebuild my conda environment a while back and since then, I am no longer running into this issue, so I assume it was something related to out-of-sync versions of various packages.

zssherman, 2023-09-26:
@kylejgillett No worries at all! And sounds good, glad to hear you got it working! I'll close the issue for now.

closed 2023-09-26 by zssherman