ARM-DOE/pyart #1431: Move submodule graph.cm to own package

kmuehlbauer, 2023-05-31:
* Py-ART version: any
* Python version: any 
* Operating System: any

### Description

I'm using the colorblind-friendly colormaps defined in that submodule in conjunction with other packages. It would be great if that submodule could be moved into a dedicated package (maybe under the openradar organization) so that it can be used with other packages without the need to install Py-Art. I think the whole radar ecosystem (and probably others) might benefit from this. Not only from better accessibility but also from better maintainability. xradar is already one good example.

If this happens it might also be highlighted in upcoming conferences and papers (ping @zssherman).

zssherman, 2023-05-31:
@kmuehlbauer Yeah, I think that's a great idea! Let me discuss with Max and we'll see how to implement this in it's own module.

zssherman, 2023-05-31:
@kmuehlbauer So we discussed having a pyart_colormaps module, that way the calls to the colormap doesn't break any workflows, but that way its still a separate module. Thoughts?

zssherman, 2023-06-06:
I'm hoping to take a crack at this early next week.

zssherman, 2023-06-27:
I promise I haven't forgotten about this, just been real busy...

zssherman, 2023-07-11:
Colormaps have been added to the openradar/cmweather https://github.com/openradar/cmweather repository, working on installation and testing next.

closed 2023-09-11 by mgrover1

kmuehlbauer, 2023-09-11:
Thanks all, who made this happen! This is a great achievement. 🎉