ARM-DOE/pyart #1432: How is the gate_altitude calculated?

chenuanc, 2023-06-01:
* Py-ART version: 1.14.5
* Python version:  python 3.8
* Operating System: windows10

I want to know how the gate_altitudes are calculated because I have found some very large values.

zssherman, 2023-06-01:
@chenuanc Gate altitudes are calculated using range azimuths and elevation angles in the radar. These values are fed intop the antenna_vectors_to_cartesion function here:
https://github.com/ARM-DOE/pyart/blob/143c7560d98e9f180855c6549855ae05e195fdd9/pyart/core/transforms.py#L81

Follows the 4/3 earth radius model and uses np.meshgrid to return gate x, y, z.
Gate altitude then is z plus the radar altitude. Hope that helps!

chenuanc, 2023-06-02:
> @chenuanc Gate altitudes are calculated using range azimuths and elevation angles in the radar. These values are fed intop the antenna_vectors_to_cartesion function here: [...]

This is very helpful! Thank you!

mgrover1, 2023-06-14:
> * Py-ART version: 1.14.5
> * Python version:  python 3.8
> * Operating System: windows10
> 
> I want to know how the gate_altitudes are calculated because I have found some very large values.

Do you have any specific datasets this was causing an issue with?