ARM-DOE/pyart #1449: I cannot save my radar data as a NetCDF4 file

December0w0, 2023-09-06:
* Py-ART version: pyart-0.2.3
* Python version: python-v3.9
* Operating System: windows11

### Description

I used the pyart package to deal with X-band phased array radar(PAR) in Guangdong, province of the People's Republic of China. while using command pyart.io.write_cfradical, it says that I cannot save my file because of permission denied and the shape of the array. However, the plotting part where using pyart ran pretty well.

### What I Did

```
### Command
`pyart.io.write_cfradial(outputdir+'cfradial_'+os.path.basename(_)+'.nc', radar)`
### Traceback1(Permission Denied)
`Traceback (most recent call last):

  File D:\Anaconda3\Lib\site-packages\spyder_kernels\py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File e:\代码\python代码\radar_qc_outputcfradial（cinrad）.py:207
    pyart.io.write_cfradial(outputdir+'cfradial_'+os.path.basename(_)+'.nc', radar)

  File D:\Anaconda3\Lib\site-packages\pyart\io\cfradial.py:489 in write_cfradial
    dataset = netCDF4.Dataset(filename, "w", format=format)

  File src\netCDF4\_netCDF4.pyx:2464 in netCDF4._netCDF4.Dataset.__init__

  File src\netCDF4\_netCDF4.pyx:2027 in netCDF4._netCDF4._ensure_nc_success

PermissionError: [Errno 13] Permission denied: 'E:/Data/XRadar_Guangdong/figure/Saola/withoutcoreRadar/outputcfradial_Z_RADR_I_ZG320_20230901030059_O_DOR_AXPT0364_CRA_FMT.bin.bz2.nc'
`
### Traceback2(array dimension)
`Traceback (most recent call last):

  File src\netCDF4\_netCDF4.pyx:5476 in netCDF4._netCDF4.Variable.__setitem__

ValueError: cannot reshape array of size 40 into shape (1,)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File D:\Anaconda3\Lib\site-packages\spyder_kernels\py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File e:\代码\python代码\radar_qc_outputcfradial（cinrad）.py:206
    pyart.io.write_cfradial(outputdir+'cfradial_'+os.path.basename(_)+'.nc', radar)

  File D:\Anaconda3\Lib\site-packages\pyart\io\cfradial.py:644 in write_cfradial
    _create_ncvar(radar.instrument_parameters[k], dataset, k, dim)

  File D:\Anaconda3\Lib\site-packages\pyart\io\cfradial.py:887 in _create_ncvar
    ncvar[:] = data[:]

  File src\netCDF4\_netCDF4.pyx:5478 in netCDF4._netCDF4.Variable.__setitem__

  File <__array_function__ internals>:200 in broadcast_to

  File D:\Anaconda3\Lib\site-packages\numpy\lib\stride_tricks.py:413 in broadcast_to
    return _broadcast_to(array, shape, subok=subok, readonly=True)

  File D:\Anaconda3\Lib\site-packages\numpy\lib\stride_tricks.py:349 in _broadcast_to
    it = np.nditer(

ValueError: input operand has more dimensions than allowed by the axis remapping
`
If there was a crash, please include the traceback here.
```

mgrover1, 2023-09-06:
@December0w0 - it appears you are using a much older version Py-ART. Can you try upgrading your version?

One other thing to check is that this file does not already exist. You can try deleting the file `E:/Data/XRadar_Guangdong/figure/Saola/withoutcoreRadar/outputcfradial_Z_RADR_I_ZG320_20230901030059_O_DOR_AXPT0364_CRA_FMT.bin.bz2.nc` if it exists, and make sure that you have write permissions to this directory.

December0w0, 2023-09-07:
> @December0w0 - it appears you are using a much older version Py-ART. Can you try upgrading your version?
>
> One other thing to check is that this file does not already exist. You can try deleting the [...]

Hey bro, I tried to update my py-ART and now it is 1.15.2. I also deleted my nc file and it does not working. However, I used the S-band radar at the same place and it ran pretty well by using Py-ART. I tried to save the nc file at the same path E:/Data/XRadar_Guangdong/figure/Saola/withoutcoreRadar/ and the S-band radar data was saved successfully but the X-band phased array radar data still cannot be saved as an nc file. Why this could be happening? Prehaps is the formation of different kind of radar data?

mgrover1, 2023-09-07:
The phased array data format likely has more dimensions than the S-band data you are using. Without the sample files, I am not sure I can help here. Can you share a small sample file?

December0w0, 2023-09-08:
> 
Sadly, I cannot give u a sample file as my tutor does not agree :(. However, I can provide the type and the format data. I know it cannot be solved without the sample, but I will try to analysis the dimensions of the array and the function in the Py-ART

mgrover1, 2023-09-08:
@December0w0 - is it okay if we transfer this into a discussion, where we can follow up more?