ARM-DOE/pyart #1481: Scan mode issue when reading cfradial

swnesbitt, 2023-10-27:
* Py-ART version: 1.15.2
* Python version: 3.11.4
* Operating System: tested on MacOS and Linux

### Description

Tried to open a cfradial 360 sweep file produced from RadxConvert (originally a DOW produced DORADE)

### What I Did

Tried to produce a radar display, but it says can't deal with `scan_mode` of "other" in the radar object.

Looked at the cfradial file, and the `sweep_mode` is `azimuth_surveillance` which is legal according to cfradial conventions.

However, pyart.io.read is reading in `scan_mode` from the cfradial as a masked array rather than a string.

```python
 radar = pyart.io.read('/Users/snesbitt/data/DOW7-231005-cfrad/20231005/cfrad.20231005_153454.950_to_20231005_153512.742_DOW7high_SUR.nc')
radar.sweep_mode
{'long_name': 'scan_mode_for_sweep',
 'options': 'sector, coplane, rhi, vertical_pointing, idle, azimuth_surveillance, elevation_surveillance, sunscan, pointing, calibration, manual_ppi, manual_rhi, sunscan_rhi, doppler_beam_swinging, complex_trajectory, electronic_steering',
 'data': masked_array(data=[[b'a', b'z', b'i', b'm', b'u', b't', b'h', b'_', b's',
                     b'u', b'r', b'v', b'e', b'i', b'l', b'l', b'a', b'n',
                     b'c', b'e', --, --, --, --, --, --, --, --, --, --,
                     --, --]],
              mask=[[False, False, False, False, False, False, False,
                     False, False, False, False, False, False, False,
                     False, False, False, False, False, False,  True,
                      True,  True,  True,  True,  True,  True,  True,
                      True,  True,  True,  True]],
        fill_value=b'',
             dtype='|S1')}
```

This causes plotting routines to crash as it compares `sweep_mode` to a number of known strings, and upon failure of matching, sets `sweep_mode` to "other".

swnesbitt, 2023-10-27:
Manually setting `radar.scan_type='ppi'` as appropriate is a workaround.

mgrover1, 2023-10-31:
@swnesbitt - can you share a sample file here? I can take a look at debugging here.

mgrover1, 2023-10-31:
@swnesbitt - I think I found the culprit here. The byte-encoded string is being converted to a list... we should add a check on the dimensionality of the sweep mode array before grabbing the first element
https://github.com/ARM-DOE/pyart/blob/main/pyart/io/cfradial.py#L200

PR #1484 (2023-11-02): FIX: Fix setting scan_mode when scan_name is empty
This fixes reading in the scan_mode when the scan_name is an empty string

- [x] Closes #1481 
- [x] Tests added
- [x] Documentation reflects changes
closed 2023-11-02 by mgrover1