ARM-DOE/pyart #1498: grid_from_radars generates misleading gridded arrays when applied on the last sweep of a given file

isilber, 2023-12-19:
* Py-ART version: 1.16.1
* Python version: 3.11.5
* Operating System: Red Hat 7.9

### Description

When converting` radar` objects to a `grid `object (following a single sweep extraction) using the `grid_from_radars` with the default `map_gates_to_grid` algorithm, gridded fields have an apparent maximum range much smaller than the true range. This artifact only occurs when the last sweep is processed, so if nsweeps is 4, the artifact will be generated only when gridding is applied to the 4th sweep.
I saw this issue in multiple files in multiple datastreams (e.g., ARM's anxkasacr and houkasacr) and this issue is consistent.
The ROI field appears consistent in all cases, while the map module does not appear to use any sweep information directly. I have no leads on this, but I suspect the solution should be straightforward (an indexing issue?) to someone more familiar with the code base.

Working example shown below.

### What I Did

```
import matplotlib.pyplot as plt
import cartopy.crs as ccrs
import pyart

# path, file, and fields
file_path = '/path/to/datastream/anxkasacrcfrcorppivM1.c1/'
file_name = 'anxkasacrcfrcorppivM1.c1.20200312.003009.nc'
processed_field = 'reflectivity_at_cor'

# load file
radar_obj = pyart.io.read(f"{file_path}{file_name}")  # load file
print(f"Total sweeps = {radar_obj.nsweeps}")

# Plot polar coordinate (raw) data
print(f"now displaying Ze data for 2nd and 4th (final) sweeps in polar coordinates\n"
      f"(both approaching the 40 km range)")
fig = plt.figure(figsize=(12,6), tight_layout=True)
fig.suptitle('polar')
for sweep, ax_ind in zip([1, 3], range(2)):
    ax = fig.add_subplot(1, 2, ax_ind + 1, projection=ccrs.Mercator())
    sweep_obj = radar_obj.extract_sweeps((sweep,))
    display = pyart.graph.RadarMapDisplay(sweep_obj)
    display.plot_ppi_map(
        processed_field,
        sweep=0,
        ax=ax,
    )
    display.plot_range_rings([10, 20, 30, 40])
plt.show()

print(f"now displaying gridded Ze data for 2nd and 4th (final) sweeps")
fig2 = plt.figure(figsize=(12,6), tight_layout=True)
fig2.suptitle('Cartesian gridded')
for sweep, ax_ind in zip([1, 3], range(2)):
    sweep_obj = radar_obj.extract_sweeps((sweep,))
    grid = pyart.map.grid_from_radars(
    (sweep_obj,),
    grid_shape=(1, 1601, 1601),
    grid_limits=((0, 10000.), [-40000, 40000], [-40000, 40000]),
    fields=[processed_field],
    )
    ax = fig2.add_subplot(1, 2, ax_ind + 1)
    ax.imshow(grid.fields[processed_field]["data"][0], origin="lower", extent=(-40, 40, -40, 40))
plt.show()
```

mgrover1, 2024-01-16:
@isilber - sorry for the delay here, do you have a sample file we can test this with?

isilber, 2024-01-16:
[anxkasacrcfrcorppivM1.c1.20200312.003009.zip](https://github.com/ARM-DOE/pyart/files/13952744/anxkasacrcfrcorppivM1.c1.20200312.003009.zip)
@mgrover1  I'm attaching the file you can use in the example I added above. Thanks!

mgrover1, 2024-01-17:
@isilber - is it okay if we transfer this over to a discussion?