ARM-DOE/pyart #1499: Slow read speeds for certain Level2 files

nawendt, 2023-12-20:
* Py-ART version: 1.16.1
* Python version: 3.11.7
* Operating System: Any

Certain Level2 files that I read with pyART are processed much slower than you would expect. Level2 files from WSR-88D sites are quite fast. However, I have some x-band data in Level2 format that takes much longer to read. The x-band has higher resolution (about 2x the gates as the 88D), but that should not mean significantly longer processing. Below is some profiling data when reading in the x-band data.

```
         531832 function calls (531814 primitive calls) in 18.385 seconds

   Ordered by: internal time
   List reduced from 366 to 10 due to restriction <10>

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        1   16.696   16.696   17.984   17.984 /opt/miniconda3/envs/radar/lib/python3.11/site-packages/pyart/io/nexrad_level2.py:587(_decompress_records)
     2881    1.283    0.000    1.283    0.000 {method 'decompress' of '_bz2.BZ2Decompressor' objects}
        6    0.107    0.018    0.132    0.022 /opt/miniconda3/envs/radar/lib/python3.11/site-packages/pyart/io/nexrad_level2.py:489(get_data)
    31687    0.085    0.000    0.097    0.000 /opt/miniconda3/envs/radar/lib/python3.11/site-packages/pyart/io/nexrad_level2.py:767(_unpack_structure)
    28800    0.038    0.000    0.176    0.000 /opt/miniconda3/envs/radar/lib/python3.11/site-packages/pyart/io/nexrad_level2.py:658(_get_msg31_data_block)
     2880    0.029    0.000    0.230    0.000 /opt/miniconda3/envs/radar/lib/python3.11/site-packages/pyart/io/nexrad_level2.py:640(_get_msg31_from_buf)
    51851    0.026    0.000    0.039    0.000 /opt/miniconda3/envs/radar/lib/python3.11/site-packages/pyart/io/nexrad_level2.py:756(_structure_size)
    83555    0.013    0.000    0.013    0.000 {method 'join' of 'str' objects}
        9    0.012    0.001    0.012    0.001 {method 'astype' of 'numpy.ndarray' objects}
    31686    0.012    0.000    0.133    0.000 /opt/miniconda3/envs/radar/lib/python3.11/site-packages/pyart/io/nexrad_level2.py:761(_unpack_from_buf)
```
Most of the time is spend decompressing. Since metPy has a Level2 reader, I was curious if it also took much longer on the x-band data. The timing results are below.

```python
# x-band using metPy
673 ms ± 2.77 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)

# x-band using pyART
16.2 s ± 932 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
```
That is a substantial difference. I had wondered if the file structure of the x-band was perhaps responsible, but with metPy reading it quickly that seems less likely. I am wondering if you all can spot what might be causing it to be so slow. I have attached an example file so you can check it out.

[test_data.zip](https://github.com/ARM-DOE/pyart/files/13729725/test_data.zip)

kmuehlbauer, 2023-12-20:
@nawendt Py-ART creates new bz2.BZ2Decompressor() objects when looping over the filestream. MetPy uses one-shot bz2.decompress() function. 

It looks like Py-ART might just re-use the above the loop  created bz2.BZ2Decompressor() object instead of recreating at each loop interval. 

https://github.com/ARM-DOE/pyart/blob/90a9b93ed63dbe544a2b5b4c7e71d39d30231f6b/pyart/io/nexrad_level2.py#L587-L601

mgrover1, 2024-01-02:
Thanks for the heads up here! We will get this fix in.

zssherman, 2024-01-03:
As I'm not too familiar with the decompressor, I found:

Note This class does not transparently handle inputs containing multiple compressed streams, unlike [decompress()](https://docs.python.org/3/library/bz2.html#bz2.decompress) and [BZ2File](https://docs.python.org/3/library/bz2.html#bz2.BZ2File). If you need to decompress a multi-stream input with [BZ2Decompressor](https://docs.python.org/3/library/bz2.html#bz2.BZ2Decompressor), you must use a new decompressor for each stream.

Seems to be the reason why we call the decompressor object again.

However I am trying to think of more efficient ways of doing this

kmuehlbauer, 2024-01-04:
@zssherman I see, that makes sense. Looking at that now, I've did overlook the obvious. Wrapping the first decopressor-call in `bytearray` should do the trick:

```python
buf = bytearray(decompressor.decompress(cbuf[skip:]))
```

It seems that the it might have to do with recreation of immutable `bytes` in the loop.

zssherman, 2024-01-04:
@kmuehlbauer Ah makes sense! Great catch! Thanks for finding that! Yeah I was digging for a bit yesterday with no luck, so this is awesome!

closed 2024-01-04 by mgrover1