ARM-DOE/pyart #1500: plot a cross section from a grid failed

jaydonshi, 2024-01-01:
* Py-ART version: 1.16.1
* Python version: 3.12.1
* Operating System: Windows 11

### Description

plot a cross section from a grid failed

example link: [https://arm-doe.github.io/pyart/examples/plotting/plot_cross_section.html](url)

[image]

### What I Did

```
import cartopy.crs as ccrs
import matplotlib.pyplot as plt

import pyart
from pyart.testing import get_test_data

# Read in the data from two XSAPR radars
[xsapr_sw_file](https://docs.python.org/3/library/stdtypes.html#str) = get_test_data("swx_20120520_0641.nc")
[xsapr_se_file](https://docs.python.org/3/library/stdtypes.html#str) = get_test_data("sex_20120520_0641.nc")
radar_sw = pyart.io.read_cfradial([xsapr_sw_file](https://docs.python.org/3/library/stdtypes.html#str))
radar_se = pyart.io.read_cfradial([xsapr_se_file](https://docs.python.org/3/library/stdtypes.html#str))

# Filter out gates with reflectivity > 100 from both radars
gatefilter_se = pyart.filters.GateFilter(radar_se)
gatefilter_se.exclude_transition()
gatefilter_se.exclude_above("corrected_reflectivity_horizontal", 100)
gatefilter_sw = pyart.filters.GateFilter(radar_sw)
gatefilter_sw.exclude_transition()
gatefilter_sw.exclude_above("corrected_reflectivity_horizontal", 100)

# perform Cartesian mapping, limit to the reflectivity field.
grid = pyart.map.grid_from_radars(
    (radar_se, radar_sw),
    gatefilters=(gatefilter_se, gatefilter_sw),
    grid_shape=(20, 181, 181),
    grid_limits=((500, 10000), (-50000, 40000), (-60000, 40000)),
    grid_origin=(36.57861, -97.363611),
    fields=["corrected_reflectivity_horizontal"],
)

# Define some start and end points, using (latitude, longitude)
[start](https://docs.python.org/3/library/stdtypes.html#tuple) = (36.7, -97.7)
[end](https://docs.python.org/3/library/stdtypes.html#tuple) = (36.2, -97.8)

# Setup the figure, and plot our x/y view of the radar
[fig](https://matplotlib.org/stable/api/figure_api.html#matplotlib.figure.Figure) = [plt.figure](https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.figure.html#matplotlib.pyplot.figure)(figsize=(18, 6))
ax1 = [plt.subplot](https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.subplot.html#matplotlib.pyplot.subplot)(121, projection=ccrs.PlateCarree())
display = pyart.graph.GridMapDisplay(grid)
display.plot_grid(
    "corrected_reflectivity_horizontal",
    ax=ax1,
    cmap="pyart_HomeyerRainbow",
    vmin=-20,
    vmax=70,
)

# Plot our start and end points, as well as a line in between the two
[ax1.scatter](https://matplotlib.org/stable/api/_as_gen/matplotlib.axes.Axes.scatter.html#matplotlib.axes.Axes.scatter)([start](https://docs.python.org/3/library/stdtypes.html#tuple)[1], [start](https://docs.python.org/3/library/stdtypes.html#tuple)[0], color="tab:blue", label="Start")
[ax1.scatter](https://matplotlib.org/stable/api/_as_gen/matplotlib.axes.Axes.scatter.html#matplotlib.axes.Axes.scatter)([end](https://docs.python.org/3/library/stdtypes.html#tuple)[1], [end](https://docs.python.org/3/library/stdtypes.html#tuple)[0], color="black", label="End")
[ax1.plot](https://matplotlib.org/stable/api/_as_gen/matplotlib.axes.Axes.plot.html#matplotlib.axes.Axes.plot)([[start](https://docs.python.org/3/library/stdtypes.html#tuple)[1], [end](https://docs.python.org/3/library/stdtypes.html#tuple)[1]], [[start](https://docs.python.org/3/library/stdtypes.html#tuple)[0], [end](https://docs.python.org/3/library/stdtypes.html#tuple)[0]], color="k", linestyle=":")
[plt.legend](https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.legend.html#matplotlib.pyplot.legend)(loc="upper right")

# Add a cross section, using our start and end points, and set our x-axis as latitude (lat)
[ax2](https://matplotlib.org/stable/api/_as_gen/matplotlib.axes.Axes.html#matplotlib.axes.Axes) = [plt.subplot](https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.subplot.html#matplotlib.pyplot.subplot)(122)
display.plot_cross_section(
    "corrected_reflectivity_horizontal",
    [start](https://docs.python.org/3/library/stdtypes.html#tuple),
    [end](https://docs.python.org/3/library/stdtypes.html#tuple),
    x_axis="lat",
    cmap="pyart_HomeyerRainbow",
    vmin=-20,
    vmax=70,
)
```

zssherman, 2024-01-01:
@jaydonshi Thanks for the info! We will take a look and see if the units registry changed and if standards need to be updated.

mgrover1, 2024-01-02:
I am taking a look at this today!

mgrover1, 2024-01-02:
This is due to some updates on the MetPy side - I added a PR over here with the fix
https://github.com/Unidata/MetPy/pull/3343

For now, please revert your MetPy version to less than 1.6.

PR #1507 (2024-01-12): FIX: Remove the metpy pin for doc build and CI
Remove the pin for metpy versions, which was added there to resolve issues with the cross section functionality.

- [x] Closes #1500 
- [x] Tests added
- [x] Documentation reflects changes
closed 2024-01-12 by mgrover1