ARM-DOE/pyart #1510: Error occured while trying to import pyart

December0w0, 2024-01-26:
* Py-ART version:1.17.0
* Python version:3.10.9
* Operating System:Windows11

### Description

I tried to import pyart today but it seems that there was some problems with pyart and numpy model. It seemed that pyart cannot use an attribute in numpy.io. My pyart version is 1.17.0 and my numpy version is 1.26.3
\

### What I Did
import pyart
upgrade and downgrade numpy
```
Traceback (most recent call last):

  File D:\Anaconda3\lib\site-packages\spyder_kernels\py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File e:\代码\课题同化代码\pycwr改进处理.py:10
    import pyart

  File D:\Anaconda3\lib\site-packages\pyart\__init__.py:29
    from . import (

  File D:\Anaconda3\lib\site-packages\pyart\graph\__init__.py:57
    from . import cm, cm_colorblind  # noqa

  File D:\Anaconda3\lib\site-packages\pyart\graph\cm_colorblind.py:18
    from ._cm_colorblind import datad

  File D:\Anaconda3\lib\site-packages\pyart\graph\_cm_colorblind.py:42
    bal_rgb_vals = np.genfromtxt(os.path.join(data_dir, "balance-rgb.txt"))

  File D:\Anaconda3\lib\site-packages\numpy\lib\npyio.py:1959 in genfromtxt
    raise ValueError("'max_rows' must be at least 1.")

  File D:\Anaconda3\lib\site-packages\numpy\lib\__init__.py:90 in __getattr__
    raise AttributeError("module {!r} has no attribute "

AttributeError: module 'numpy.lib' has no attribute '_datasource'
```

zssherman, 2024-01-26:
@December0w0 Hmm interesting. Yeah I was looking at the numpy issue board and didn't see anything. Your versions are up to date as well. I see your using anaconda, are you using environments? I would almost say maybe try a fresh environment if you haven't already? Could be an older pip numpy or something else breaking things.

mgrover1, 2024-01-26:
@December0w0 what are you please try
```bash
conda uninstall arm_pyart
conda install arm_pyart
```

This will ensure the proper dependencies are installed, I suspect something odd is going on with the upgrading/downgrading numpy step.

December0w0, 2024-01-27:
> @December0w0 Hmm interesting. Yeah I was looking at the numpy issue board and didn't see anything. Your versions are up to date as well. I see your using anaconda, are you using environments? I [...]

Ok, I'll try it

December0w0, 2024-01-27:
> @December0w0 what are you please try
> 
> ```shell
> conda uninstall arm_pyart
> conda install arm_pyart
> ```
> 
> This will ensure the proper dependencies are installed, I suspect something odd is going on with the upgrading/downgrading numpy step.

Understood. I used conda install to install pyart. Prehaps pip might work?

December0w0, 2024-01-27:
> @December0w0 what are you please try
> 
> ```shell
> conda uninstall arm_pyart
> conda install arm_pyart
> ```
> 
> This will ensure the proper dependencies are installed, I suspect something odd is going on with the upgrading/downgrading numpy step.

Thx. It worked. I also updated my anaonda3 and reinstalled pyart. But the problem is that another package failed and something odd happened between it and numpy. I will try to downgrade my numpy and hope it might work.

December0w0, 2024-01-27:
> Understood. I used conda install to install pyart. Prehaps pip might work?

Problem solved. Downgrade numpy and reinstall my pyart did work. Thanks. I'll close my issue.

December0w0, 2024-01-27:
> @December0w0 Hmm interesting. Yeah I was looking at the numpy issue board and didn't see anything. Your versions are up to date as well. I see your using anaconda, are you using environments? I [...]

Thx, It worked. My pyart version now is 1.15.0 and my numpy version is 1.24.3. Thanks a lot. I'll close this issue.

closed 2024-01-27 by December0w0