ARM-DOE/pyart #1522: Incorporation of RadTraQ into Py-ART

AdamTheisen, 2024-03-06:
### Description

RadTraQ was initially developed for quality monitoring of vertically pointing radar data from the ARM program.  With the Py-ART move to xradar, it seems like it could be a good time to incorporate the remaining codes in RadTraQ into Py-ART as they are based on xarray.  There is some overlap already on the CFAD and column extraction but the remaining functions in RadTraQ could be worthwhile to pull in.

https://github.com/ARM-Development/RadTraQ

The end-goal would be to retire RadTraQ and focus future development efforts in Py-ART.

rcjackson, 2024-08-28:
@mgrover1 I will be working on integrating some of the RadTraQ features for PyART 2.0 this afternoon/tomorrow.

zssherman, 2024-09-20:
Think I accidently closed this. Still more in Radtraq that needs to be brought over.

PR #1637 (2024-08-30): ADD: Add corner reflector example
Add a corner reflector example

- [x] Closes #1522 
- [x] Tests added
- [x] Documentation reflects changes
closed 2025-05-13 by zssherman