ARM-DOE/pyart #1531: Get the beam width using the aux_io.read_sinarame_h5 function

evmpython, 2024-03-19:
* Py-ART version: 1.18.0
* Python version: 3.10.13
* Operating System: Ubuntu 22.04.3 LTS

### Description
Hi all. I am trying extract the beam width from HDF5 from Braziliam radar using the function “aux_io.read_sinarame_h5”. But the “instruments_parameters” is empty.  I am using Google Colab. 

When the files is open with hdfview software, the radar file have this information (see below). How Can I get the  beam width in this file? Follow one HDF5 radar file as example. 

[2024021622400300dBZ.vol.zip](https://github.com/ARM-DOE/pyart/files/14652868/2024021622400300dBZ.vol.zip)

### What I Did

!pip install -q condacolab
import condacolab
condacolab.install()
!mamba install -q -c conda-forge arm_pyart

import pyart  
filename = '2024021622400300dBZ.vol.h5'
radar = pyart.aux_io.read_sinarame_h5(f'/content/{filename}')
radar.instrument_parameters.keys()

AttributeError                            Traceback (most recent call last)
[<ipython-input-1-9d68e307ffea>](https://localhost:8080/#) in <cell line: 9>()
      7 radar = pyart.aux_io.read_sinarame_h5(f'/content/{filename}')
      8 
----> 9 radar.instrument_parameters.keys()

AttributeError: 'NoneType' object has no attribute 'keys'
[image]

zssherman, 2024-03-19:
@evmpython The sinararme reading in py-art at the moment is in aux_io, so intrument parameters defaults to none as it does not currently have a method in the code that pulls it. These files i'm familiar with, but if its store in some way in h5py, we can always add that in.

evmpython, 2024-03-19:
Also i have done test with pyart.aux_io.read_odim_h5. But the same error persist.

closed 2024-03-20 by mgrover1

mrugna, 2024-08-21:
Sorry for reopening this issue :)

I created a fresh install using 1.18.7 and found this modification so I'll try to bring a little bit of context.

This sinarame function was made a long time ago as a separate version of the ODIM H5 (aux) reader because it was meant to support the files created from argentine radar hardware ("sinarame" kind of stands for Argentina national radar network). This had to be done in a separate function because of some missing parameters in the H5 file which is not complete or somewhat compliant.

Specifically to this issue, I don't recognize this file but I see from the screenshot that it might be a Rainbow radar so definitely not a sinarame radar.

So now I'm finding this error because the sinarame files don't have this attribute on the main object. I'm happy to provide a sample file in the near future just in case.

```
File "/home/mrugna/mambaforge/envs/py312_kdp/lib/python3.12/site-packages/pyart/aux_io/sinarame_h5.py", line 352, in read_sinarame_h5
    if len(hfile["how"].attrs) == 0:
KeyError: "Unable to synchronously open object (object 'how' doesn't exist)"
```

I'll try to make an enhancement _soon_ to catch this on a try-except instead of len that gives this key error.

PR #1532 (2024-03-20): ENH: Add instrument parameter check for sinarame.
Note: I'm not sure if this is the most efficient or best way for this, so can see what others have to add.

- [x] Closes #1531 
- [x] Documentation reflects changes