ARM-DOE/pyart #1546: Not be8ing able to write_grid_mdv properly

MohitDahliya, 2024-04-05:
* Py-ART version: 1.16.1
* Python version: 3.11.5
* Operating System: Windows

### Description

I wanted to convert nc radr files into grid_mmdv files
While the conversion did happen but the filed of reflectivity was not there in the newly created grid_mdv file, only a few fields were there. I was expecting that all the files in the grid file would come into the mdv file

### What I Did

import numpy as np
import pyart

radar = pyart.io.read('DLI190207055229-IMD-B.nc')
radar.fields.keys()
out >> dict_keys(['T', 'Z', 'V', 'W', 'ZDR', 'KDP', 'PHIDP', 'SQI', 'RHOHV', 'HCLASS'])

grid = pyart.map.grid_from_radars(radar, grid_shape=(41, 401, 401),
    grid_limits=((0000.0, 20000.0), (-200000.0, 200000.0), (-200000.0, 200000.0)),
    gridding_algo='map_gates_to_grid', copy_field_dtypes=True)
grid.fields.keys()
out >> dict_keys(['HCLASS', 'ZDR', 'W', 'SQI', 'PHIDP', 'RHOHV', 'T', 'Z', 'V', 'KDP', 'ROI'])

pyart.io.write_grid_mdv('output_grid.mdv', grid, mdv_field_names= None, field_write_order= None)

file = pyart.io.read_grid_mdv("output_grid.mdv")

file.fields.keys()
out >> dict_keys(['differential_reflectivity', 'cross_correlation_ratio', 'differential_phase', 'specific_differential_phase'])

mgrover1, 2024-04-05:
@MohitDahliya - thanks for raising this issue, by default the mdv writer will only write the fields with mappings to mdv data

`
mdv_field_names (dict or None, optional) – Mapping between grid fields and MDV data type names. Field names mapped to None or with no mapping will be excluded from writing. If None, the same field names will be used.
`

https://arm-doe.github.io/pyart/API/generated/pyart.io.write_grid_mdv.html#pyart.io.write_grid_mdv

You would need to provide these mappings to write out the additional fields

MohitDahliya, 2024-04-06:
How could I provide these mappings ?

mgrover1, 2024-04-09:
@MohitDahliya Sorry for the delay - I did not work on this during the weekend, and was attending the solar eclipse yesterday. We will take a look at this this week.

MohitDahliya, 2024-04-09:
Okay.. Appreciate your response..

mgrover1, 2024-04-10:
@MohitDahliya - did that solution @zssherman work for you?

mgrover1, 2024-04-16:
@MohitDahliya - we are following up here, did that solution work?

MohitDahliya, 2024-04-24:
yes, it worked. I checked default_config.py and found that I needed to change the name of my reflectivity field from 'Z' to 'DBZ'. However, I think you people should change this fixed file name and instead focus on the long name of the field as not everyone would have the name of the reflectivity field as 'DBZ' in their radar file.

MohitDahliya, 2024-04-24:
@mgrover1 I saw that you have also handled some issues related to CSU_Radartools. I am also facing some issues in running that code. Can you help me with that ?

mgrover1, 2024-04-24:
> yes, it worked. I checked default_config.py and found that I needed to change the name of my reflectivity field from 'Z' to 'DBZ'. However, I think you people should change this fixed file name and [...]

Pull requests are always welcome! 

> @mgrover1 I saw that you have also handled some issues related to CSU_Radartools. I am also facing some issues in running that code. Can you help me with that ?

I would suggest opening an issue on the repository describing your issue. I am on travel this week, and likely would not be able to take a look until next week. One of the other developers there should be able to help.

closed 2024-04-24 by mgrover1