ARM-DOE/pyart #1553: Missing get_sweep_keys from xradar

waarmond, 2024-04-10:
* Py-ART version: git master
* Python version: 3.10
* Operating System: Linux from scratch
* [xradar](https://github.com/openradar/xradar): 0.3.0/0.5.0

### Description

`ImportError: cannot import name 'get_sweep_keys' from 'xradar.util' (/usr/lib/python3.10/site-packages/xradar/util.py)`

### What I Did

```
import pyart
```

zssherman, 2024-04-10:
@waarmond get_sweep_keys was added fairly recently to xradar. Your xradar is 0.5.0 on your system? My guess was maybe an outdated xradar.

Wait I see you mention 0.5.0 above.

Hmm, will have to look into it

waarmond, 2024-04-10:
@zssherman My error. I've started the python instance before updating xradar.

mgrover1, 2024-04-10:
@waarmond - is it working with the updated version?

zssherman, 2024-04-10:
@mgrover1 Sounds good, thanks!

mgrover1, 2024-04-30:
@waarmond - following up here, is it working?

mgrover1, 2024-06-21:
Closing this with the new release

closed 2024-06-21 by mgrover1