ARM-DOE/pyart #1561: ENH: Update documentation of `hydroclass_semisupervised` for X band classification.

RBhupi, 2024-04-15:
* Py-ART version: All
* Python version: All
* Operating System: All

### Description
In `hydroclass_semisupervised` if the frequency information in `radar.instrument_parameters` is missing, function defaults to C band and warns radar frequency is unknown. The documentation is not clear on how to get hydrometeor class for X band in this situation. Need for adding band information to radar.instrument_parameters should be clearly stated, may be with an example.

### What I Did
- [ ] Update documentation to clarify that classification for X band needs frequency information in the radar object or it defaults to C band.
- [ ] Give example for correctly adding X band information to radar.instrument_parameters.

RBhupi, 2024-04-15:
Example of warning

```
# Get classification
hydromet_class = pyart.retrieve.hydroclass_semisupervised(
    radar, 
    refl_field="corrected_reflectivity",
    zdr_field="corrected_differential_reflectivity",
    kdp_field="filtered_corrected_specific_diff_phase",
    rhv_field="RHOHV",
    temp_field="sounding_temperature",
)
```

> /Users/bhupendra/anaconda3/envs/data/lib/python3.12/site-packages/pyart/retrieve/echo_class.py:663: UserWarning: Radar frequency unknown. Default coefficients for C band will be applied.

zssherman, 2024-04-15:
Thanks for the discussion! Would you be willing to do a PR to expand on the warning etc for it?

RBhupi, 2024-04-15:
Thanks @zssherman ! I have sent a PR for documentation update.

PR #1563 (2024-04-16): ENH:Docs:hydroclass
- [x] Closes #1561
- [x] Documentation reflects changes
closed 2024-04-16 by mgrover1