ARM-DOE/pyart #1568: Probability of hail using pyart

MohitDahliya, 2024-04-24:
* Py-ART version: 1.17.0
* Python version: 3.11.5
* Operating System: Windows

### Description

How can I create a spatial plot of 'probability of hail' (POH) using pyart? There are some algorithms available such as TITAN which can compute POH, but can I create a similar code using pyart ?

### What I Did

Haven't really done any programming yet for this purpose. Hoping to get a headstart on how to begin.

zssherman, 2024-04-24:
I dont believe there currently is an exact code in Py-ART that calculates that, however I believe PyHail might which uses Py-ART as a dependency, you could always give that a look. Can also see if anyone else states anything here

scollis, 2024-04-24:
Hey Mohit this is more a radar science question. Suggest you visit the open radar discourse https://openradar.discourse.group/t/welcome-to-open-radar-discourse/7

MohitDahliya, 2024-04-25:
@zssherman I am working with PyHail currently. Seems like a useful package. Thanks for the heads up. Let's see how it progress.

MohitDahliya, 2024-04-25:
#run pipeline for a single tilt (may not be the lowest sweep)
import time
t0 = time.time()
pipeline(radar_fn, 2, close_fig=False)    
t1 = time.time()
print('elapsed time:', t1-t0, 's')

This should have given the plots of MESH and Accumulated hail, but instead it gives
file failed:  <pyart.core.radar.Radar object at 0x0000023D14BF8810>
elapsed time: 0.0 s

why has the file failed ? I don't understand.

mgrover1, 2024-04-25:
Without seeing your `pipeline` function, we are not able to troubleshoot here - if you are using PyHail, I encourage you to open an issue there with a detailed code snippet

MohitDahliya, 2024-04-25:
Okay, thanks for the reply.

mgrover1, 2024-04-25:
Closing since this is more of a discussion

closed 2024-04-25 by mgrover1