ARM-DOE/pyart #1584: Level3 product with code 153 is not supported

SquarerFive, 2024-05-21:
* Py-ART version:
 arm_pyart 1.18.3 py311h459d7ec_0
* Python version:
Python 3.11.9
* Operating System:
Ubuntu 20.04 (WSL)

### Description

I have tried loading a NEXRAD radar product (N0B) using pyart from the public unidata-nexrad AWS repo but got an error:
```
Level3 product with code 153 is not supported
```

I assume this is a new format that is not being processed by pyart yet?

### What I Did

```
pyart.io.nexradl3_read.read_nexrad_level3('s3://unidata-nexrad-level3/GRR_N0B_2024_05_20_23_58_47')
```

It would be great to get some clarity on this issue.

Thanks!
Aiden.

zssherman, 2024-05-21:
@SquarerFive Sadly the nexrad level 3 reader predates my experience with that code base. If I had to take a guess:
https://github.com/ARM-DOE/pyart/issues/311 "the NEXRADLevel3 class has only implemented support for reading radial Level 3 data products and raises the NotImplementedError for non-radial data products" so it could be that it does not play well with the Py-ART radar class. Whether its a newer format, that i'm not too sure. I would have to see when the 153 super resolution products were created https://www.roc.noaa.gov/wsr88d/PublicDocs/NewTechnology/2620001XICDforRPGtoClassIUser.pdf

We can also see if anyone else has ideas as well. Few of the other Py-ART folks are currently away this week at the moment

dopplershift, 2024-05-21:
@zssherman The 153 (and other products) I think have started to be distributed in the last couple of years. This is a pretty needed update because the previous N0Q, etc. products have been almost completely discontinued in favor of N0B and friends. I'm 95% percent sure the previous code you used for N0Q should work with 153/N0B. MetPy looks like it's using the same code paths for both.

@SquarerFive If you have a pressing need, MetPy has support for these products, though the interface is MUCH rougher to deal with than PyART's.

zssherman, 2024-05-21:
@dopplershift Thanks for the reply! This is extremely helpful! Yeah I can play around with our current reader and see if I can get these files read in. I also second your point that, @SquarerFive you could use metpy in the mean time, while we take a look at this.

zssherman, 2024-05-29:
[image]

looks like we are getting somewhere, just still unsure on the resolution for the three msg products.

dopplershift, 2024-05-30:
@zssherman We're setting a max range of 460 km (~248 nmi) in MetPy. The super resolution products are all on half-degree azimuth resolution. Somewhere they also changed ref to use 250m / 0.13 nmi range resolution, but that might have already been present in the previous N0Q family of products.

zssherman, 2024-05-30:
@dopplershift Makes sense, thank you!

PR #1591 (2024-06-03): ENH: Update nexrad level 3 reader to read super resolution data.
Initial PR might need to be updated more.

- [x] Closes #1584