ARM-DOE/pyart #1588: Grid.projection being str not handled

juhi24, 2024-05-28:
* Py-ART version: f789a7f6b5aa1bf6b86666a25c4595bc13d56446
* Python version: any
* Operating System: any

### Description

Docstring of `pyart.core.grid.Grid` mentions that the attribute `projection` could be either a `dict` or a `str`. However, in `Grid.get_projparams`, Grid.projection is expected to be `dict` and the case for a `str` is not handled at all.

Either support for `str` should not be implied in the documentation or it should be fully supported or there should be disclaimers and proper error messages about partial support.

zssherman, 2024-05-28:
@juhi24 The fix I put in should fix the issue, thanks again for letting us know!

closed 2024-05-28 by zssherman

juhi24, 2024-05-29:
Awesome! That was fast, thanks!

PR #1589 (2024-05-28): FIX: Update pyproj functions to take string.

- [x] Closes #1588 
- [x] Tests added
- [x] Documentation reflects changes