ARM-DOE/pyart #1590: FR: Export radar file after edit

MaxMyzer, 2024-05-29:
### Description
Add a feature to allow the program to export radar files after manipulating them.
Came here after suggesting something similar in [metpy](https://github.com/Unidata/MetPy/issues/3515)

Note - I am currently working on a program to do something similar, but having it built into a library such as this would be good. 

### Specification 
- The program should accept (arbitrary) modifications to an input file
  - Things like the AVSET, VCP, and what sweeps are included
- The program should write radar files in the correct format (For level 2 files, see ICDs `2620002` and `2620010`)

### Why?
- Example 1: Some radar programs only process full volumes, so SAILS sweeps are lost. Using this, one could spoof each SAILS sweep into its own "volume"
- Example 2: Simulating scenarios - rather than relying solely on existing data, using simulated data can test the limits of experimental algorithms.

mgrover1, 2024-05-29:
This would be possible with editing the fields in some radar object, then exporting to a cfradial file

```python
import pyart

# Read the data
radar = pyart.io.read("some_file.nc")

# Apply your corrections/retrievals
radar.correct...

# Write the file
pyart.io.write_cfradial(radar, "new_file.nc")
```

Is there something missing here?

MaxMyzer, 2024-06-02:
> Is there something missing here?

I mainly want to do this with a Level2 file, although being able to do it with cfradical is awesome

mgrover1, 2024-06-03:
Would exporting to cfradial allow you accomplish your task? We do not plan to support Py-ART writing to NEXRAD formats unfortunately, especially with the World Meteorological Organization deciding upon the Cfradial2/FM301 data standard.

MaxMyzer, 2024-06-05:
I see, I will look into that.

closed 2024-06-05 by MaxMyzer