ARM-DOE/pyart #1599: Handling missing metadata fields in pyart.aux_io.read_odim_h5() reader

jorahu, 2024-06-21:
* Py-ART version: 1.18.3
* Python version: 3.10.1
* Operating System: Linux

### Description
Importing h5 file with pyart.aux_io.read_odim_h5() could not complete due to missing "version" or "source" field in file metadata. 
The problem is with the lines 169 and 170 in https://github.com/ARM-DOE/pyart/blob/main/pyart/aux_io/odim_h5.py
There should be a check like there is for example lines 177 and 178.

### What I Did
First, line 164 should be deleted as it is duplicated at line 170.

The lines 168-170 could be changed as follows:
h_what = hfile["what"].attrs 
if "version" in h_what:
    metadata["version"] = _to_str(h_what["version"])
if "source" in h_what:
    metadata["source"] = _to_str(h_what["source"])

I am not sure how this complies with ODIM structure, but I tested it with 2 completely different files (from different parts of the world), one had missing "source", other had missing "version" for unknown reasons. One missing (**potentially** unimportant) metadata field should not stop the whole reading process.

mgrover1, 2024-06-21:
@jorahu - any interest in submitting a PR with the fix here? Thanks for highlighting the current limitation.

Would you be interested in supplying those sample files so we can add that to the test suite?

jorahu, 2024-06-21:
Yeah, see... can't even comment without messing it up. I'm pretty new to github. Apologies.

mgrover1, 2024-06-21:
No worries! If you could share some of the files to test, that would be ideal, to make sure that the fix works generically!

Feel free to share it with my email, mgrover@anl.gov

PR #1749 (2025-05-27): MNT: Update to check for metadata in ODIM reader

- [x] Closes #1599
closed 2025-05-27 by mgrover1