ARM-DOE/pyart #1602: BUG: Cannot install arm-pyart 1.18.5 on macOS (arm64)

neutrinoceros, 2024-06-25:
* Py-ART version: 1.18.5
* Python version: 3.12.4
* Operating System: macOS

### Description

Trying to instal the package in a fresh venv on this arch errors as
```shell
$ python -m pip install arm-pyart
...
ERROR: arm-pyart has an invalid wheel, arm-pyart has an invalid wheel, could not read 'arm_pyart-1.18.5.dist-info/WHEEL' file: BadZipFile("Bad CRC-32 for file 'arm_pyart-1.18.5.dist-info/WHEEL'")
```

version 1.18.4 still installs correctly

mgrover1, 2024-06-25:
@neutrinoceros - the conda-forge wheels pass, as well as installing locally on my Mac... do you have more info about the machine that is having issues?

neutrinoceros, 2024-06-25:
I discovered this in yt's CI ([example logs](https://github.com/yt-project/yt/actions/runs/9659702514/job/26644579993)), but I was also able to reproduce locally.
In case that's useful, here are my specs:
- macOS 14.5
- chip: M2 Pro

I tried it with pip 24.1, 24.0 and 23.3.2, and the result was identical every time.
I also checked with `pip install --no-cache-dir ...`, and `uv` (0.2.15):
```
❯ uv pip install arm-pyart
Resolved 56 packages in 853ms
error: Failed to prepare distributions
  Caused by: Failed to fetch wheel: arm-pyart==1.18.5
  Caused by: Failed to extract archive
  Caused by: deflate decompression error
```

neutrinoceros, 2024-06-25:
Also: 
- installing from sdist with `--no-binary arm-pyart` works
- note that I'm reproducing on Python *3.12*. This seems actually crucial; I don't have any problem for 3.11 or 3.10

mgrover1, 2024-06-25:
That makes more sense why I can't reproduce locally (I have an M1 Pro chip) - I am curious if this line in the cibuildwheel configuration could be to blame?
https://github.com/ARM-DOE/pyart/blob/main/.github/workflows/wheels.yml#L59C29-L59C52

neutrinoceros, 2024-06-25:
I doubt it. I don't think it changes the behavior of anything if binaries for your dependencies are available anyway.
But it might well be a bug in cibuildwheel itself, or its tool chain. I don't have time left to look it up today, but I might try to escalate this tomorrow (of course, feel free to beat me to it).

mgrover1, 2024-06-25:
I appreciate the feedback here @neutrinoceros - the issue has been created, hopefully we can sort this out today!
https://github.com/pypa/cibuildwheel/issues/1896

mgrover1, 2024-06-25:
This should resolve the issue https://github.com/ARM-DOE/pyart/pull/1606

We were overwriting the built objects

mgrover1, 2024-06-25:
A new version is being cut after tests run + pass +merge

closed 2024-06-25 by zssherman

mgrover1, 2024-06-25:
@neutrinoceros - can you please try rerunning your CI with the new wheels released in the latest version?

neutrinoceros, 2024-06-25:
Done. All good. Thanks a lot !

mgrover1, 2024-06-25:
Thank you! Keeping this closed!

PR #1606 (2024-06-25): FIX: Ensure apple builds on its own systems
Ensure apple builds are not duplicated + overwriting each other

- [x] Closes #1602 
- [x] Tests added
- [x] Documentation reflects changes