ARM-DOE/pyart #1608: Errors with Updated NumPy2.0 in Py-ART

RBhupi, 2024-07-08:
* Py-ART version: 1.18.6 , build py312h9a8786e_0  from  conda-forge
* Python version: 3.12
* Operating System: Linux 4.18.0-513.9.1.el8_9.x86_64

### Description
After updating NumPy2.0, following error is raised due to the removal of `np.unicode_` while loading the pyart packages. 

### What I Did
1. create environment and install NumPy 2.0.
2. Import Py-ART in a JuPyter Notebook.
3. Got the Error .

```
{
	"name": "AttributeError",
	"message": "`np.unicode_` was removed in the NumPy 2.0 release. Use `np.str_` instead.",
	"stack": "---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[1], line 4
      1 #import xarray as xr
      2 #import xradar as xd
      3 import glob
----> 4 import pyart
      5 import pandas as pd
      7 import matplotlib.pyplot as plt

File ~/data-env1/lib/python3.12/site-packages/pyart/__init__.py:29
     25 import importlib.metadata as _importlib_metadata
     27 # import subpackages
     28 # print out helpful message if build fails or importing from source tree
---> 29 from . import (
     30     __check_build,  # noqa
     31     aux_io,  # noqa
     32     bridge,  # noqa
     33     config,  # noqa
     34     core,  # noqa
     35     correct,  # noqa
     36     filters,  # noqa
     37     graph,  # noqa
     38     io,  # noqa
     39     map,  # noqa
     40     retrieve,  # noqa
     41     testing,  # noqa
     42     util,  # noqa
     43     xradar,  # noqa
     44 )
     45 from ._debug_info import _debug_info  # noqa
     47 # root level functions

File ~/data-env1/lib/python3.12/site-packages/pyart/aux_io/__init__.py:16
      1 \"\"\"
      2 Additional classes and functions for reading and writing data from a number
      3 of file formats.
   (...)
     13 
     14 \"\"\"
---> 16 from .arm_vpt import read_kazr  # noqa
     17 from .d3r_gcpex_nc import read_d3r_gcpex_nc  # noqa
     18 from .edge_netcdf import read_edge_netcdf  # noqa

File ~/data-env1/lib/python3.12/site-packages/pyart/aux_io/arm_vpt.py:13
     10 import numpy as np
     12 from ..config import FileMetadata
---> 13 from ..core.radar import Radar
     14 from ..io import cfradial
     17 def read_kazr(
     18     filename,
     19     field_names=None,
   (...)
     23     include_fields=None,
     24 ):

File ~/data-env1/lib/python3.12/site-packages/pyart/core/__init__.py:6
      1 \"\"\"
      2 Core Py-ART classes and function for interacting with weather radar data.
      3 
      4 \"\"\"
----> 6 from .grid import Grid  # noqa
      7 from .radar import Radar  # noqa
      8 from .radar_spectra import RadarSpectra  # noqa

File ~/data-env1/lib/python3.12/site-packages/pyart/core/grid.py:10
      7 from netCDF4 import num2date
      9 try:
---> 10     import xarray
     12     _XARRAY_AVAILABLE = True
     13 except ImportError:

File ~/data-env1/lib/python3.12/site-packages/xarray/__init__.py:1
----> 1 from xarray import testing, tutorial
      2 from xarray.backends.api import (
      3     load_dataarray,
      4     load_dataset,
   (...)
      8     save_mfdataset,
      9 )
     10 from xarray.backends.zarr import open_zarr

File ~/data-env1/lib/python3.12/site-packages/xarray/testing.py:10
      7 import numpy as np
      8 import pandas as pd
---> 10 from xarray.core import duck_array_ops, formatting, utils
     11 from xarray.core.dataarray import DataArray
     12 from xarray.core.dataset import Dataset

File ~/data-env1/lib/python3.12/site-packages/xarray/core/duck_array_ops.py:36
     33 from numpy.core.multiarray import normalize_axis_index  # type: ignore[attr-defined]
     34 from numpy.lib.stride_tricks import sliding_window_view  # noqa
---> 36 from xarray.core import dask_array_ops, dtypes, nputils
     37 from xarray.core.parallelcompat import get_chunked_array_type, is_chunked_array
     38 from xarray.core.pycompat import array_type, is_duck_dask_array

File ~/data-env1/lib/python3.12/site-packages/xarray/core/dask_array_ops.py:3
      1 from __future__ import annotations
----> 3 from xarray.core import dtypes, nputils
      6 def dask_rolling_wrapper(moving_func, a, window, min_count=None, axis=-1):
      7     \"\"\"Wrapper to apply bottleneck moving window funcs on dask arrays\"\"\"

File ~/data-env1/lib/python3.12/site-packages/xarray/core/dtypes.py:43
     33 NINF = AlwaysLessThan()
     36 # Pairs of types that, if both found, should be promoted to object dtype
     37 # instead of following NumPy's own type-promotion rules. These type promotion
     38 # rules match pandas instead. For reference, see the NumPy type hierarchy:
     39 # https://numpy.org/doc/stable/reference/arrays.scalars.html
     40 PROMOTE_TO_OBJECT: tuple[tuple[type[np.generic], type[np.generic]], ...] = (
     41     (np.number, np.character),  # numpy promotes to character
     42     (np.bool_, np.character),  # numpy promotes to character
---> 43     (np.bytes_, np.unicode_),  # numpy promotes to unicode
     44 )
     47 def maybe_promote(dtype):
     48     \"\"\"Simpler equivalent of pandas.core.common._maybe_promote
     49 
     50     Parameters
   (...)
     57     fill_value : Valid missing value for the promoted dtype.
     58     \"\"\"

File ~/data-env1/lib/python3.12/site-packages/numpy/__init__.py:397, in __getattr__(attr)
    394     raise AttributeError(__former_attrs__[attr])
    396 if attr in __expired_attributes__:
--> 397     raise AttributeError(
    398         f\"`np.{attr}` was removed in the NumPy 2.0 release. \"
    399         f\"{__expired_attributes__[attr]}\"
    400     )
    402 if attr == \"chararray\":
    403     warnings.warn(
    404         \"`np.chararray` is deprecated and will be removed from \"
    405         \"the main namespace in the future. Use an array with a string \"
    406         \"or bytes dtype instead.\", DeprecationWarning, stacklevel=2)

AttributeError: `np.unicode_` was removed in the NumPy 2.0 release. Use `np.str_` instead."
}
```

zssherman, 2024-07-08:
@RBhupi What xarray version do you have? Looks like the error might be coming from an outdated xarray version

mgrover1, 2024-07-16:
@RBhupi - following up here, which version of xarray are you using?

RBhupi, 2024-07-17:
xarray                    2023.6.0        py312h06a4308_0

mgrover1, 2024-07-18:
@RBhupi - you need to update your version of xarray to work with the latest version of numpy

mgrover1, 2024-08-19:
@RBhupi - did updating your version of xarray resolve the issue?

RBhupi, 2024-08-21:
I was trying to update xarray earlier but that was still installing older version, now after I updated the conda itself, the numpy vrsion is downgraded. But I guess that will work now.

```

(/ccsopen/home/braut/data-env1) bash-4.4$ conda list numpy
# packages in environment at /ccsopen/home/braut/data-env1:
#
# Name                    Version                   Build  Channel
numpy                     2.0.0           py312h22e1c76_0    conda-forge
(/ccsopen/home/braut/data-env1) bash-4.4$ conda list xarray
# packages in environment at /ccsopen/home/braut/data-env1:
#
# Name                    Version                   Build  Channel
xarray                    2023.6.0        py312h06a4308_0  
xarray-datatree           0.0.14             pyhd8ed1ab_0    conda-forge
(/ccsopen/home/braut/data-env1) bash-4.4$ conda update xarray
Collecting package metadata (current_repodata.json): - /sw/cirrus/python/3.7/anaconda-base/lib/python3.7/site-packages/cryptography/hazmat/backends/openssl/x509.py:17: CryptographyDeprecationWarning: This version of cryptography contains a temporary pyOpenSSL fallback path. Upgrade pyOpenSSL now.
  utils.DeprecatedIn35,
done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 4.11.0
  latest version: 24.7.1

Please update conda by running

    $ conda update -n base -c defaults conda

## Package Plan ##

  environment location: /ccsopen/home/braut/data-env1

  added / updated specs:
    - xarray

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    blas-1.0                   |         openblas          46 KB
    bokeh-3.4.1                |  py312he106c6f_0         5.7 MB
    cftime-1.6.2               |  py312ha883a20_1         233 KB
    charset-normalizer-3.3.2   |     pyhd3eb1b0_0          44 KB
    cloudpickle-3.0.0          |  py312h06a4308_0          40 KB
    contourpy-1.2.0            |  py312hdb19cb5_0         238 KB
    matplotlib-base-3.8.4      |  py312h526ad5a_0         7.6 MB
    numpy-1.26.4               |  py312h2809609_0          10 KB
    numpy-base-1.26.4          |  py312he1a6c75_0         7.7 MB
    openjpeg-2.5.2             |       he7f1fd0_0         371 KB
    pip-24.2                   |  py312h06a4308_0         2.8 MB
    requests-2.32.3            |  py312h06a4308_0         123 KB
    setuptools-72.1.0          |  py312h06a4308_0         2.9 MB
    snappy-1.2.1               |       h6a678d5_0          46 KB
    statsmodels-0.14.2         |  py312ha883a20_0        12.2 MB
    ------------------------------------------------------------
                                           Total:        40.0 MB

The following NEW packages will be INSTALLED:

  blas               pkgs/main/linux-64::blas-1.0-openblas
  numpy-base         pkgs/main/linux-64::numpy-base-1.26.4-py312he1a6c75_0

The following packages will be UPDATED:

  charset-normalizer                     2.0.4-pyhd3eb1b0_0 --> 3.3.2-pyhd3eb1b0_0
  cloudpickle                         2.2.1-py312h06a4308_0 --> 3.0.0-py312h06a4308_0
  pip                                  24.0-py312h06a4308_0 --> 24.2-py312h06a4308_0
  requests                           2.32.2-py312h06a4308_0 --> 2.32.3-py312h06a4308_0
  setuptools                         69.5.1-py312h06a4308_0 --> 72.1.0-py312h06a4308_0
[... 134 lines cut ...]
snappy-1.2.1         | 46 KB     | ###################################################################################### | 100% 
blas-1.0             | 46 KB     | ###################################################################################### | 100% 
requests-2.32.3      | 123 KB    | ###################################################################################### | 100% 
charset-normalizer-3 | 44 KB     | ###################################################################################### | 100% 
bokeh-3.4.1          | 5.7 MB    | ###################################################################################### | 100% 
pip-24.2             | 2.8 MB    | ###################################################################################### | 100% 
Preparing transaction: done
Verifying transaction: done
Executing transaction: done
```
```
(/ccsopen/home/braut/data-env1) bash-4.4$ conda list xarray
# packages in environment at /ccsopen/home/braut/data-env1:
#
# Name                    Version                   Build  Channel
xarray                    2023.6.0        py312h06a4308_0  
xarray-datatree           0.0.14             pyhd8ed1ab_0    conda-forge
(/ccsopen/home/braut/data-env1) bash-4.4$ conda list numpy
# packages in environment at /ccsopen/home/braut/data-env1:
#
# Name                    Version                   Build  Channel
numpy                     1.26.4          py312h2809609_0  
numpy-base                1.26.4          py312he1a6c75_0  
(/ccsopen/home/braut/data-env1) bash-4.4$ 
```

closed 2024-08-26 by RBhupi