ARM-DOE/pyart #1691: Data masking issue when interpolating to Grid

lauratomkins, 2024-11-18:
* Py-ART version: 1.19.2
* Python version: 3.9.20
* Operating System: Windows 10

### Description

I'm working on interpolating some tropical cyclone data and noticed that the data seems to get masked after interpolating to a Grid object. This is also evident in some of the cases that I put together for [this example](https://arm-doe.github.io/pyart/examples/retrieve/plot_feature_detection.html)
[image]

I'm not sure what's going on here and haven't been able to dig much into it, but it seems a little similar to issues #957 and #942 . I was able to replicate the masking and fuzzy edges with both nearest neighbor and cressman weighting methods.  

This can be replicated with the examples using NEXRAD data on [this page](https://arm-doe.github.io/pyart/examples/retrieve/plot_feature_detection.html). Considering the non-NEXRAD examples are working as expected, could this be something that changed with the NEXRAD level2 files? It seems unlikely to me, but I can't see anything that changed in the gridding algorithm. Any insight would be appreciated!

zssherman, 2024-11-18:
Hi @lauratomkins , what is the current ROI set as? Looking at the bottom right, looks like some missing data that is being picked up by the interpolation. Going to add in @rcjackson as well if he has insights has he was in this discussion as well before. But my guess would be the ROI.

lauratomkins, 2024-11-18:
@zssherman , it looks like it is an issue with the ROI. I have been using the default values but there were some changes to these default ROI values 7 months ago #1552.   
Specific line [here](https://github.com/ARM-DOE/pyart/blame/473318d3bbb4041a9a705f5524e18c51908d2db4/pyart/map/grid_mapper.py#L298)

When I change `nb` back to the original value of 1.5, I get much a much more reasonable interpolation and no fuzzy/artifacts at the edges.

Interpolation with (new) default `nb` of 1.0
[image]

Interpolation with previous default `nb` of 1.5
[image]

Should I update the function calls in my example page to explicitly set the `nb` values to 1.5? Would it be a possibility to have a dynamic `nb` value based on the source of the input data? Or at the least perhaps an example page or a blog post that explains the ROI process/variables and lists some best practices for interpolating data from different radars? I can imagine as a first time Py-ART user working with NEXRAD data and the default parameters, this output would be pretty confusing.

closed 2024-11-18 by lauratomkins