ARM-DOE/pyart #1692: PyART - module not found error on JUpyter Notebook

gkb999, 2024-11-19:
* Py-ART version: 
* Python version: 3.12
* Operating System: Ubuntu

### Module Not found error
I'm trying to use basic input/output functions from https://arm-doe.github.io/pyart/examples/io/plot_older_nexrad_data_aws.html#sphx-glr-examples-io-plot-older-nexrad-data-aws-py 

When I import the basic packages: 
```import cartopy.crs as ccrs
import matplotlib.pyplot as plt

import pyart
```

it says 'cartopy' - module not found and the same with 'pyART'.

### I even cross checked within my package,
```
Requirement already satisfied: arm_pyart in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (1.19.3)
Requirement already satisfied: numpy in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (1.26.4)
Requirement already satisfied: scipy in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (1.14.1)
Requirement already satisfied: netCDF4 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (1.6.4)
Requirement already satisfied: matplotlib in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (3.9.2)
Requirement already satisfied: pooch in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (1.8.2)
Requirement already satisfied: cftime in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (1.6.4)
Requirement already satisfied: fsspec in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (2024.10.0)
Requirement already satisfied: s3fs in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (2024.10.0)
Requirement already satisfied: open-radar-data in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (0.4.1)
Requirement already satisfied: xradar>=0.8.0 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (0.8.0)
Requirement already satisfied: pandas in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (2.2.2)
Requirement already satisfied: mda-xdrlib in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (0.2.0)
Requirement already satisfied: xarray>=2024.10.0 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (2024.10.0)
Requirement already satisfied: cartopy in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (0.24.0)
Requirement already satisfied: pint in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from arm_pyart) (0.24.4)
Requirement already satisfied: packaging>=23.1 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from xarray>=2024.10.0->arm_pyart) (24.2)
Requirement already satisfied: python-dateutil>=2.8.2 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from pandas->arm_pyart) (2.9.0)
Requirement already satisfied: pytz>=2020.1 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from pandas->arm_pyart) (2024.2)
Requirement already satisfied: tzdata>=2022.7 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from pandas->arm_pyart) (2024.2)
Requirement already satisfied: cmweather in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from xradar>=0.8.0->arm_pyart) (0.3.2)
Requirement already satisfied: dask in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from xradar>=0.8.0->arm_pyart) (2024.11.2)
Requirement already satisfied: h5netcdf>=1.0.0 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from xradar>=0.8.0->arm_pyart) (1.4.1)
Requirement already satisfied: h5py>=3.0.0 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from xradar>=0.8.0->arm_pyart) (3.12.1)
Requirement already satisfied: lat-lon-parser in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from xradar>=0.8.0->arm_pyart) (1.3.1)
Requirement already satisfied: pyproj in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from xradar>=0.8.0->arm_pyart) (3.7.0)
Requirement already satisfied: xmltodict in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from xradar>=0.8.0->arm_pyart) (0.14.2)
Requirement already satisfied: shapely>=1.8 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from cartopy->arm_pyart) (2.0.6)
Requirement already satisfied: pyshp>=2.3 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from cartopy->arm_pyart) (2.3.1)
Requirement already satisfied: contourpy>=1.0.1 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from matplotlib->arm_pyart) (1.3.1)
Requirement already satisfied: cycler>=0.10 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from matplotlib->arm_pyart) (0.12.1)
Requirement already satisfied: fonttools>=4.22.0 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from matplotlib->arm_pyart) (4.55.0)
Requirement already satisfied: kiwisolver>=1.3.1 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from matplotlib->arm_pyart) (1.4.7)
Requirement already satisfied: pillow>=8 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from matplotlib->arm_pyart) (11.0.0)
Requirement already satisfied: pyparsing>=2.3.1 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from matplotlib->arm_pyart) (3.2.0)
Requirement already satisfied: certifi in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from netCDF4->arm_pyart) (2024.8.30)
Requirement already satisfied: platformdirs>=2.1.0 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from pint->arm_pyart) (4.3.6)
Requirement already satisfied: typing-extensions>=4.0.0 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from pint->arm_pyart) (4.12.2)
Requirement already satisfied: flexcache>=0.3 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from pint->arm_pyart) (0.3)
Requirement already satisfied: flexparser>=0.4 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from pint->arm_pyart) (0.4)
Requirement already satisfied: requests>=2.19.0 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from pooch->arm_pyart) (2.32.3)
Requirement already satisfied: aiobotocore<3.0.0,>=2.5.4 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from s3fs->arm_pyart) (2.15.2)
Requirement already satisfied: aiohttp!=4.0.0a0,!=4.0.0a1 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from s3fs->arm_pyart) (3.11.2)
Requirement already satisfied: botocore<1.35.37,>=1.35.16 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from aiobotocore<3.0.0,>=2.5.4->s3fs->arm_pyart) (1.35.36)
Requirement already satisfied: wrapt<2.0.0,>=1.10.10 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from aiobotocore<3.0.0,>=2.5.4->s3fs->arm_pyart) (1.16.0)
Requirement already satisfied: aioitertools<1.0.0,>=0.5.1 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from aiobotocore<3.0.0,>=2.5.4->s3fs->arm_pyart) (0.12.0)
Requirement already satisfied: aiohappyeyeballs>=2.3.0 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from aiohttp!=4.0.0a0,!=4.0.0a1->s3fs->arm_pyart) (2.4.3)
Requirement already satisfied: aiosignal>=1.1.2 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from aiohttp!=4.0.0a0,!=4.0.0a1->s3fs->arm_pyart) (1.3.1)
Requirement already satisfied: attrs>=17.3.0 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from aiohttp!=4.0.0a0,!=4.0.0a1->s3fs->arm_pyart) (24.2.0)
Requirement already satisfied: frozenlist>=1.1.1 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from aiohttp!=4.0.0a0,!=4.0.0a1->s3fs->arm_pyart) (1.5.0)
Requirement already satisfied: multidict<7.0,>=4.5 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from aiohttp!=4.0.0a0,!=4.0.0a1->s3fs->arm_pyart) (6.1.0)
Requirement already satisfied: propcache>=0.2.0 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from aiohttp!=4.0.0a0,!=4.0.0a1->s3fs->arm_pyart) (0.2.0)
Requirement already satisfied: yarl<2.0,>=1.17.0 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from aiohttp!=4.0.0a0,!=4.0.0a1->s3fs->arm_pyart) (1.17.1)
Requirement already satisfied: six>=1.5 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from python-dateutil>=2.8.2->pandas->arm_pyart) (1.16.0)
Requirement already satisfied: charset-normalizer<4,>=2 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from requests>=2.19.0->pooch->arm_pyart) (3.4.0)
Requirement already satisfied: idna<4,>=2.5 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from requests>=2.19.0->pooch->arm_pyart) (3.10)
[... cut ...]
Requirement already satisfied: toolz>=0.10.0 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from dask->xradar>=0.8.0->arm_pyart) (1.0.0)
Requirement already satisfied: jmespath<2.0.0,>=0.7.1 in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from botocore<1.35.37,>=1.35.16->aiobotocore<3.0.0,>=2.5.4->s3fs->arm_pyart) (1.0.1)
Requirement already satisfied: locket in ./anaconda3/envs/pyart_env/lib/python3.12/site-packages (from partd>=1.4.0->dask->xradar>=0.8.0->arm_pyart) (1.0.0)

```
It shows that both cartopy and pyart are installed.
What could be the error?

gkb999, 2024-11-19:
Okay, I opened jupyter notebook from the same environment, so presumed that wasn't error. But it was the error.
Solved now, thank you so much.

closed 2024-11-19 by gkb999

zssherman, 2024-11-19:
@gkb999 Glad to hear that it is solved! Cheers