ARM-DOE/pyart #1694: LiDAR Examples(?)

wreed1989, 2024-11-22:
I'm currently working on a project where I'm working with raw LiDAR data and I've come across a few examples like the images found here: https://www.arm.gov/news/data/post/81368 and I'm trying to figure out how to get PyART to read the data files which are HPL (.hpl). From what I've seen on the Github page/example gallery, it looks like this can be done but it's geared towards radar data rather than LiDAR data and perhaps I've been looking at it too long or something but I can't figure out how to recreate this plotting technique for my data.

This is the closest example I can find but the input is a NetCDF file so unless I can somehow convert my .hpl file to .nc, I'm not sure if this option will work. 

https://arm-doe.github.io/pyart/examples/plotting/plot_rhi_cfradial_singlescan.html#sphx-glr-examples-plotting-plot-rhi-cfradial-singlescan-py

Here is the head of my data for a visual:

Filename:	User1_100_20240714_122137.hpl
System ID:	100
Number of gates:	833
Range gate length (m):	12.0
Gate length (pts):	4
Pulses/ray:	3000
No. of waypoints in file:	1
Scan type:	User file 1 - csm
Focus range:	65535
Start time:	20240714 12:21:45.80
Resolution (m/s):	0.0382
Range of measurement (center of gate) = (range gate + 0.5) * Gate length
Data line 1: Decimal time (hours)  Azimuth (degrees)  Elevation (degrees) Pitch (degrees) Roll (degrees)
f9.6,1x,f6.2,1x,f6.2
Data line 2: Range Gate  Doppler (m/s)  Intensity (SNR + 1)  Beta (m-1 sr-1) Spectral Width
i3,1x,f6.4,1x,f8.6,1x,e12.6,1x,f6.4 - repeat for no. gates
**** Instrument spectral width = 0.305763
12.36262778  90.00  -0.00 -0.46 -1.77
  0 0.1529 1.339292  1.910213E-5 0.0382 
  1 -0.6115 1.239853  1.350899E-5 0.0382 
  2 -3.0576 1.074272  4.185253E-6 0.0382 
  3 -14.1033 1.056070  3.161522E-6 0.0382 
  4 2.4079 1.034895  1.969022E-6 0.0382 
  5 2.7519 1.015216  8.592881E-7 16.3583 
  6 3.0959 1.038732  2.189486E-6 18.2311 
  7 1.4524 1.035328  1.999168E-6 29.4297 
  8 2.5990 1.022900  1.297460E-6 13.1478 
  9 6.8797 1.036261  2.057073E-6 20.6773 
 10 1.6053 1.027125  1.540954E-6 19.6453 

Any guidance would be greatly appreciated.

zssherman, 2024-11-22:
@wreed1989 Hello! Yeah I believe the lidar is specific to netcdf only. However, I believe xradar, which we will be using for our readers eventually, reads in hpl files:
https://docs.openradarscience.org/projects/xradar/en/stable/notebooks/HaloPhotonics.html

You could try reading it in with xradar and use Py-ART's for plotting kinda similar to here:
https://arm-doe.github.io/pyart/examples/xradar/plot_xradar.html#sphx-glr-examples-xradar-plot-xradar-py

wreed1989, 2024-11-23:
Thanks! I'll give that a shot and see if I can get it to work for me.

closed 2026-10-28 by zssherman