ARM-DOE/pyart #1745: Missing platform motion variables

jcdehart, 2025-04-28:
### Package/OS Details 
- pyart: 1.16.1 and 2.0.1
- python: 3.12
- OS: Linux (not sure of details as its our supercomputer)
- note: issue was duplicated by @rcjackson on his system

### Issue
We needed to correct the Doppler velocity for platform motion, but we noticed that the platform motion variables/metadata were not visible in the radar object (using `vars(radar)`). I tried loading the file with both `pyart.io.read_cfradial(file)` and `pyart.io.read(file)`. 

The variables in question are shown in the ncdump screenshot below, but are eastward_velocity, northward_velocity, vertical_velocity, heading_change_rate, and pitch_change_rate. I admit I don't know whether these conform to standard variable naming conventions for these fields, but I was surprised to not see them in the object at all and thus not written to the output file at the end of my code.

[image]

Let me know if any other information is required and I'm happy to share a sample file through google drive.

Thanks!
Jen

zssherman, 2025-04-28:
Hi @jcdehart, Thanks for raising the issue! I do not see those variable names in Py-ART's default config. Either they were never added or weren't added due to not conforming to standards (this one im not sure, maybe they are standard are were just missed). When reading in the file, if you set file_field_names to True it should hopefully show up for now

zssherman, 2025-04-29:
@jcdehart I did find those variables in the cfradial documentation. I'll add those into Py-ART

zssherman, 2025-04-29:
Nvm I found it in an old 2018 document, but can't seem to find it in the cfradial standard name url last updated in March https://cfconventions.org/Data/cf-standard-names/current/build/cf-standard-name-table.html

zssherman, 2025-04-29:
@mgrover1 Do we know if these are old standard names?

jcdehart, 2025-04-30:
I reached out to Mike Dixon who provided an updated CfRadial document from 2021, which shows the three platform motion variables (eastward, northward, vertical...) in Section 5.6. 

He doesn't think there are standard names for heading_change_rate and pitch_change_rate, so I'll reach out to Michael Bell and the SEAPOL team to see if they had a reference for giving those variables those names.

[CfRadialDoc.v1.5.20211201 (1).pdf](https://github.com/user-attachments/files/19981066/CfRadialDoc.v1.5.20211201.1.pdf)

zssherman, 2025-04-30:
@jcdehart Thanks for reaching out to them! I'll add a PR with the current variables in the 2021 document you shared. I can add the others once Michael Bell responds.

jcdehart, 2025-05-30:
My apologies for the delay. I've reached back out to Mike and Michael for clarification - I see both heading_change_rate and heading_rate in the 2021 CfRadial document and I'm not sure which is preferred.

zssherman, 2025-05-30:
@jcdehart Do you by chance have the standard names for each of those?

zssherman, 2025-07-10:
@jcdehart No worries and sounds good! I'll get those last few variables added in. Thanks!

zssherman, 2025-07-10:
PR is #1777

closed 2025-07-15 by zssherman