ARM-DOE/pyart #1752: BUG: pyart.correct.despeckle_field

DavidDufton, 2025-05-16:
**Describe the bug**
I've noticed what I think is unexpected behaviour in `pyart.correct.despeckle_field` when a threshold is set. The returned gatefilter only excludes gates that exceed the threshold (or are within the bounds) but are from labelled data smaller than the size set. I would have thought that gates outside the threshold would also be excluded (though this can be done with additional gatefilter settings). 
See the example code below:

**To Reproduce**
```

raddisp = pyart.graph.RadarMapDisplay(radar)

# Run despeckle field with a threshold and size set but no additional gatefilter 
sp_gatefilter0 = pyart.correct.despeckle_field(radar,'reflectivity',threshold=0,size=50,)

# Set up a gatefilter to show expected behaviour
or_gatefilter1 = pyart.filters.GateFilter(radar)
or_gatefilter1.exclude_below('reflectivity',0)

# Second gatefilter to modify with new despeckle
or_gatefilter2 = pyart.filters.GateFilter(radar)
or_gatefilter2.exclude_below('reflectivity',0)
sp_gatefilter2 = pyart.correct.despeckle_field(radar,'reflectivity',threshold=0,size=50,gatefilter=or_gatefilter2)

# Generate example figure:
plt.figure(figsize=(13,11))
plt.subplot(221)
raddisp.plot_ppi('reflectivity',sweep=4,title='No filter')

plt.subplot(222)
raddisp.plot_ppi('reflectivity',sweep=4,gatefilter=sp_gatefilter0, title='Only speckle filter (fixed?)')

plt.subplot(223)
raddisp.plot_ppi('reflectivity',sweep=4,gatefilter=or_gatefilter1,title='Only gate limit filter')

plt.subplot(224)
raddisp.plot_ppi('reflectivity',sweep=4,gatefilter=sp_gatefilter2, title='Gate limit then speckle (expected)')
plt.show()
```

[image]

In the top right panel you can see the despeckle only removes small regions that exceed the threshold of 0dBZ and are smaller than 50 gates, I'd expect the behaviour to be that gates below the threshold would also be excluded by the returned filter.

To modify the code to get the expected output I've changed line 136 of despeckle.py from this:
```  mask_filter = gatefilter.gate_excluded```

to

``` mask_filter = np.logical_or(gatefilter.gate_excluded, labels.mask)```

which appears to work as I'd expect.

I'd be happy to try a pull request if you agree that the function isn't quite working as intended?

rcjackson, 2025-05-21:
Hello @DavidDufton! The threshold parameter in PyART's despeckle function shows the minimum threshold for detecting objects to remove and is not the minimum data analysis threshold for the entire scan. Therefore, despeckle is working as intended here, and should be combined with a gatefilter parameter to filter out gates with reflectivity below a certain threshold like you did in your bottom right plot.

Despeckling is typically done after doing a gate limit filter (like in the bottom left) to remove the left over speckles that your gatefilter didn't catch. So the code for your bottom right plot is the proper way to do a filter where you specify a minimum reflectivity threshold for the entire scan then despeckle.

DavidDufton, 2025-05-22:
Hi @rcjackson 

Thanks for the reply. Now that I know how it functions it's easy to use with that extra step being required. I do wonder if the documentation could be updated to make it clear that the threshold isn't applied to the returned gatefilter, only the despeckling, especially in the case where no prior gatefilter is input and the function returns its own, new gatefilter.

closed 2025-05-22 by DavidDufton