ARM-DOE/pyart #1760: TypeError when reading ODIM H5 files and possible solution

ejt4m8, 2025-05-23:
pyart: 1.19.1
python: 3.12.5
OS: Rocky8 Linux

Hello,

I tried reading an ODIM H5 file from a small x-band radar and received the following error:
`>>> radar = pt.aux_io.read_odim_h5(path)`
`Traceback (most recent call last):`
  `File "<stdin>", line 1, in <module>`
  `File "/home/ejt4m8/anaconda3/envs/dissert/lib/python3.12/site-packages/pyart/aux_io/odim_h5.py", line 370, in read_odim_h5`
    `fdata = np.ma.zeros((total_rays, max_nbins), dtype="float32")`
  `File "/home/ejt4m8/anaconda3/envs/dissert/lib/python3.12/site-packages/numpy/ma/core.py", line 8837, in __call__`
    `result = self._func.__call__(*args, **params).view(MaskedArray)`
`TypeError: 'numpy.float64' object cannot be interpreted as an integer`

Upon investigating my file, I discovered that "nbins" in "/dataset1/where" was indeed 801.0, a float. I actually fixed this on my own by going to the odim_h5.py file in my pyart environment and changed line 249. Originally, the line was:
`all_sweeps_nbins = [hfile[d]["where"].attrs["nbins"] for d in datasets]`
and I changed it to:
`all_sweeps_nbins = [int(hfile[d]["where"].attrs["nbins"]) for d in datasets]`
and I am now able to read my ODIM H5 files without error.  You can find an example file attached to reproduce the error.

I still wanted to bring this to your attention even though I was able to find a solution.
Thank you

zssherman, 2025-05-27:
Hi @ejt4m8 , thanks for raising the issue. Yeah I think adding in a check for it would be good. Did you want to create a PR with that fix you mentioned above?

zssherman, 2026-10-28:
Closing, addressed in #1913

closed 2026-10-28 by zssherman