ARM-DOE/pyart #1768: io.read_grid for generic NetCDF files?

adelmoral2, 2025-06-11:
Hi all!

While trying to open a gridded file with PyArt, that was originally created with LROSE Radx2Grid, I realized that it would only work with PyArt-gridded data. 

I ended up working with XARRAY, since we can't turn off those reserved variables.

Would it be possible to work on a version that can also deal with generic NetCDF in the future, using cf-attributes?

Thanks!

zssherman, 2025-06-11:
Hi @adelmoral2  What errors are you running into? It should work with generic netcdf files that are in cfradial format, what are the variables called in the dataset you are using?

adelmoral2, 2025-06-11:
@zssherman I have the same issue:
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
File [~/miniforge3/envs/pyart_env_n/lib/python3.13/site-packages/pyart/aux_io/radx_grid.py:81](http://localhost:8888/~/miniforge3/envs/pyart_env_n/lib/python3.13/site-packages/pyart/aux_io/radx_grid.py#line=80), in read_radx_grid(filename, exclude_fields, **kwargs)
     80 try:
---> 81     origin_latitude = _ncvar_to_dict(dset.variables["lat0"])
     82     # set data single value from Grid info origin from radx file

KeyError: 'lat0'

During handling of the above exception, another exception occurred:

AttributeError                            Traceback (most recent call last)
Cell In[13], line 1
----> 1 grid=pyart.aux_io.read_radx_grid('ncf_20230706_122559.nc')

File [~/miniforge3/envs/pyart_env_n/lib/python3.13/site-packages/pyart/aux_io/radx_grid.py:100](http://localhost:8888/~/miniforge3/envs/pyart_env_n/lib/python3.13/site-packages/pyart/aux_io/radx_grid.py#line=99), in read_radx_grid(filename, exclude_fields, **kwargs)
     97 except KeyError:
     98     origin_latitude = get_metadata("latitude")
     99     origin_latitude["data"] = [
--> 100         dset.variables["grid_mapping_0"].latitude_of_projection_origin
    101     ]
    103     origin_longitude = get_metadata("longitude")
    104     try:

File src[/netCDF4/_netCDF4.pyx:5030](http://localhost:8888/netCDF4/_netCDF4.pyx#line=5029), in netCDF4._netCDF4.Variable.__getattr__()

File src[/netCDF4/_netCDF4.pyx:4752](http://localhost:8888/netCDF4/_netCDF4.pyx#line=4751), in netCDF4._netCDF4.Variable.getncattr()

File src[/netCDF4/_netCDF4.pyx:1667](http://localhost:8888/netCDF4/_netCDF4.pyx#line=1666), in netCDF4._netCDF4._get_att()

File src[/netCDF4/_netCDF4.pyx:2164](http://localhost:8888/netCDF4/_netCDF4.pyx#line=2163), in netCDF4._netCDF4._ensure_nc_success()

AttributeError: NetCDF: Attribute not found

zssherman, 2025-06-12:
@adelmoral2 I'm trying to think of a solution to this. I see the grid Cartesian coords shared above are:

x0:standard_name = "longitude" ;
x0:long_name = "longitude" ;
x0:units = "degrees_east" ;
x0:axis = "X" ;
float y0(y0) ;
y0:standard_name = "latitude" ;
y0:long_name = "latitude" ;
y0:units = "degrees_north" ;
y0:axis = "Y" ;

These should be in meters and not degrees and could also be adding to the issues with the reader.