ARM-DOE/pyart #1769: Region-Based Dealiasing is unfolding correctly some sweeps, but not others.

adelmoral2, 2025-06-11:
I have been using the region-based dealisasing feature to unfold data from a project, but I encounter some errors with different sweeps.
Note that I am not adding any extra steps as I wanted to process a bulk of data first to have a sense how this was working.

I am attaching the pictures for two cases: 

Sweep 1- Before and after:

[image]
[image]

Sweep 2- Before and after:

[image]
[image]

adelmoral2, 2025-06-13:
@mgrover1 here is the test data file. This is an RHI with three sweeps. Sweep 0 and 2 work fine, but not 1: [cfrad.20250201_043626.929_to_20250201_043735.664_XPOL-MARSHALL_RHI.nc](url)

And the simple code I use. 

````
radar = pyart.io.read_cfradial(cfrad.20250201_043626.929_to_20250201_043735.664_XPOL-MARSHALL_RHI.nc)

nyq = radar.instrument_parameters["nyquist_velocity"]["data"][0]
velocity_dealiased = pyart.correct.dealias_region_based(
    radar, vel_field="VEL", nyquist_vel=nyq, centered=True)
radar.add_field("corrected_velocity", velocity_dealiased, replace_existing=True)

mgrover1, 2025-06-23:
@adelmoral2 digging into this today - will reach out with any more questions/follow up! Thank you for raising this issue!

mgrover1, 2025-06-23:
@adelmoral2 - can you attach the file here or email it to me at mgrover@anl.gov? I can't seem to download it.

mgrover1, 2025-06-23:
@adelmoral2 - I would suggest increasing your `interval_splits` parameter to at least five here! We can add some additional documentation here - but given the higher vertical resolution of rhi scans, sometimes the region based dealiasing needs additional segments to gather context and prevent from inconsistent behvaior;

The region-based dealiasing algorithm handles each sweep differently, and does not check for consistency across, which is a bit different than tools such as unravel!

We can add an additional example in our gallery highlighting how to tune some of these parameters if that sounds like a reasonable solution? Here is the code snippet I used to get things to work as expected!

```python
radar = pyart.io.read_cfradial("cfrad.20250201_043626.929_to_20250201_043735.664_XPOL-MARSHALL_RHI.nc")

nyq = radar.instrument_parameters["nyquist_velocity"]["data"][0]
velocity_dealiased = pyart.correct.dealias_region_based(
    radar, vel_field="VEL", interval_splits=5, nyquist_vel=nyq, centered=True)
radar.add_field("corrected_velocity", velocity_dealiased, replace_existing=True)
```

Which resulted in this corrected 270 degree scan corrected velocity field!
[image]

PR #1775 (2025-06-24): DOC: Add example for dealiasing rhi data
Add an example for dealiasing RHI data with the region-based algorithm

- [x] Closes #1769 
- [x] Tests added
- [x] Documentation reflects changes
closed 2025-06-24 by zssherman